Ribbit-Network/ribbit-network-frog-hardware #4: Create a Power Budget for the System to Size the Battery and Solar Panel

keenanjohnson, 2021-06-06:


eaudiffred, 2021-08-19:
Are you wanting to add hardware to each unit and have the power consumption reported with the air quality data?  Or just set up a test with one of them to get an estimate?

This [INA219 board](https://www.adafruit.com/product/904) looks like it could be added to each unit to measure real time power consumption.  Connecting it seems easy, but idk how you get the data, I'm out of my league there.  [(video)
](https://www.youtube.com/watch?v=BgShCD7xT_A)
If its just a test on one unit, can you sacrifice a power cord and use a multimeter?

keenanjohnson, 2021-08-20:
I just wanted to test one of them at first to get a rough estimate for emissions calculations. 

Yeah I just received an outlet power meter today so I'll run that test shortly. In the future, we'll definitely include onboard power monitoring as build out a solar-powered version of the Frog.

keenanjohnson, 2021-08-20:
See the Solar Powered Project for more tasks outstanding there :)

https://github.com/Ribbit-Network/ribbit-network-frog-sensor/projects/2

keenanjohnson, 2021-08-20:
It looks like the sensor is using about 2.6 Watts of power.

closed 2021-08-20 by keenanjohnson