Ribbit-Network/ribbit-network-frog-hardware #6: Create a Public Dashboard for the CO2 Data

keenanjohnson, 2021-06-06:


lalligagger, 2021-06-11:
I'm always looking for cool Dash projects :)

keenanjohnson, 2021-06-11:
@lalligagger! Awesome! Want to take a stab at mocking up what a Dash dashboard might be like for the sensors?

lalligagger, 2021-06-13:
I'm in. Can you drop some data in the format you're expecting? Influx DB stash is good. I looked at the public network you mentioned but their data doesn't look to be fully live yet.

keenanjohnson, 2021-06-14:
Sure thing! Here is a dump of some of the data I have in my influx DB testing database so far: 
[2021-06-13-22-45_influxdb_data.csv](https://github.com/keenanjohnson/ghg-gas-cloud/files/6645867/2021-06-13-22-45_influxdb_data.csv)

keenanjohnson, 2021-06-24:
In #9 I decided to stick with InfluxDB + Dash for the database side. 

Now that that is decided and the GPS data is being logged to the database per #10, it should be possible to create a map visualization of where each sensor is and what the current readings are at that location.

keenanjohnson, 2021-06-24:
Something like this:

https://user-images.githubusercontent.com/2559382/122119420-037eaf80-cdde-11eb-8ed7-de7aa96c0001.png

keenanjohnson, 2021-06-24:
[image]

lalligagger, 2021-06-24:
Cool, this is definitely possible! I have been slammed with another project and paper due, but I can help out at some point in the next couple of months.

Are you planning on an AWS account attached to the project? For public deployment of the dashboard, I'd be looking at Elastic Beanstalk based on experience. Heroku is another (free) option but I'm not familiar with that deployment.

keenanjohnson, 2021-06-24:
No worries @lalligagger :). This project isn't in a rush, I'm just excited about it.

Right now my goal is to keep costs down in the short term, so I'll probably start with Heroku in the short term since it is free. Seems like it won't be a big deal to transfer the hosting if AWS seems like a better long term solution.

keenanjohnson, 2021-06-24:
@JohannesBrand any thoughts on this?

keenanjohnson, 2021-06-25:
I prototyped a quick Dash dashboard hosted on Heroku in another repo [here.](https://github.com/keenanjohnson/ghg-cloud-dash)

That site is deployed to Heroku and is [viewable live at this link for now.](https://ghg-cloud.herokuapp.com/)

Just a pretty simple graph of my one sensor for now. Next will be to add a map view that can display multiple sensors. 

[image]

JohannesBrand, 2021-06-25:
> @JohannesBrand any thoughts on this?

I've used grafana and I think it's really easy to use: even a public facing dashboard was just minutes as long as you had access to a registered domain. I've looked at dash before, but never really used it. But I've seen you've been able to pull a dashboard off with dash in no time -> so it looks like a very good solution to me. great!  

one downside that I found about grafana was: it was not as customizable as I thought, so maybe another plus point for dash? However, the out-of-the-box query builder and visualizations were just great. I think that's where grafana might offer a little bit more.

lalligagger, 2021-06-25:
Wow, awesome start! This is exactly what I would have been able to tee up for you quickly, minus a Heroku deploy!
My initial suggestions on filtering would be one or more of:
- A date range selector on the timeseries plot. This could generate mean on the map and other stats in a table.
- A separate date range picker + drop-down to filters on sites, data types as well.
- Drop in a DataTable with filters turned on, for regex style filtering across columns (that can be linked to plot callback).

I could still help build in callbacks or other features as desired.

keenanjohnson, 2021-06-26:
Yeah I'd love any help as I'm still very much a dash noob haha. Those are all great suggestions though!

lalligagger, 2021-06-29:
Here is little a multi-page app w/ map I started working on a few weeks ago and then stalled on. There is also single-page version that doesn't include a map.  https://github.com/lalligagger/mtbtlm

I like the 3d global style you're using, especially for the CO2 use case! Basically, if you can narrow in on any specific features/ cross-filtering you'd like to enable, I could jump in to assist with that in mind.

keenanjohnson, 2021-07-29:
Calling this dashboard done for now and I'll create issues for more specific changes as we go.

closed 2021-07-29 by keenanjohnson