Ribbit-Network/ribbit-network-frog-hardware #9: Consider Switching the Database to Thingsboard

keenanjohnson, 2021-06-06:
https://thingsboard.io/

keenanjohnson, 2021-06-09:
Thanks for the suggestions @phil-d-wilson!

I'm currently using Influx for logging all the time series data and I'm looking for a good solution to create a nice public-facing dashboard that would allow access to the data and probably have some sort of nice map display.

I'll look into the other options and let you know! Certainly I'll reach out if I hit any balena roadblocks :)

phil-d-wilson, 2021-06-09:
Maybe the [dashboard balenaBlock](https://github.com/balenablocks/dashboard) with the [worldmap plugin](https://grafana.com/grafana/plugins/grafana-worldmap-panel/) ?

keenanjohnson, 2021-06-09:
Yeah, it seems like the standard thing would be InfluxDB -> Grafana -> World Map Plugin. I'd like a centralized database to store all the data from multiple Balena devices in the central database. I was hoping for a solution that wouldn't involve two tools (InfluxDB + Grafana) to keep cost down, but it might be the best solution.

lalligagger, 2021-06-11:
+1 for influx which I've hooked up to Dash with no problem (not a Grafana fan, gives me flashbacks to TVAC testing 😅)

keenanjohnson, 2021-06-11:
Cool yeah I haven't really looked at Dash yet, but I'll check that out.

keenanjohnson, 2021-06-24:
Ok it seems like InfluxDB + Dash for visualization is the best option for now, so going to proceed with that :)

closed 2021-06-24 by keenanjohnson