Ribbit-Network/ribbit-network-frog-hardware #11: Life Cycle Analysis for Bill of Materials

keenanjohnson, 2021-06-11:
We should do a somewhat good life cycle analysis of the materials used to construct the sensor. This should help answer questions like: What should the enclosure be made of? What is the impact of deploying X number of sensors. etc.

kevinjmiller2, 2021-07-21:
@keenanjohnson, some initial thoughts [here](https://docs.google.com/document/d/1PLEBLrZIj_m_csX-auDr22nOP_rRxhiYsVmOVrjhuzY/edit?usp=sharing) regarding lifecycle emissions per sensor. It's a very rough estimate of things, but I think it'll be a good start. Two areas where I could use some input:

1. I need to dig into the cloud part in more detail but point me toward any resources you have to help me understand how much cloud compute power you're likely using per sensor.
2. Do you have data for weight of the full system? This will be helpful for a rough scope 3 emissions estimate.

keenanjohnson, 2021-07-21:
@kevinjmiller2 ! This is incredible work! I'll review this today, but thank you so much!

1. I believe there are some good calculators from places like Microsoft for cloud compute, but we can look into that. 
2. I can get you a rough weight estimate yeah. I should add that to the BOM!

keenanjohnson, 2021-07-29:
[I added a mass column to the BOM @kevinjmiller2!](https://docs.google.com/spreadsheets/d/1lsWCXFawsJNGc44bqFXWkpjHfIFZQEj5UceJV4AiuK0/edit#gid=0)

[image]

Total mass is around 346 grams.

keenanjohnson, 2021-08-20:
In #4 I measured the power consumption of the sensor to be 2.6 Watts on average.

keenanjohnson, 2021-11-12:
@all-contributors please add @kevinjmiller2 for his great ideas and content

keenanjohnson, 2022-01-27:
I think this was an excellent first draft of an analysis for the Frog V1. We'll open another issue to complete a similar and hopefully more in depth analysis for the Frog V2 Sensor.

closed 2022-01-27 by keenanjohnson