Ribbit-Network/ribbit-network-frog-hardware #13: Evaluate whether there is a good methane sensor option

keenanjohnson, 2021-06-11:
It's been suggested that this project would be really awesome if it had both a CO2 and a Methan sensor.

We should evaluate the methane sensor options and select one if there is a suitable, fairly inexpensive sensor.

lalligagger, 2021-06-11:
What is the working principal of the CO2 sensor? I know of some optical architectures that could do both, but probably not be as cheap.

keenanjohnson, 2021-06-11:
Right now I'm working with the SCD30 sensor. It's a NDIR detection method.

https://www.sensirion.com/en/environmental-sensors/carbon-dioxide-sensors/carbon-dioxide-sensors-scd30/

keenanjohnson, 2021-06-11:
Would definitely appreciate any of your insights on what the best sensing package is to use to balance accuracy and cost :)

keenanjohnson, 2021-08-11:
It just really seems like there isn't a good methane sensor at a reasonable price, so I am going to push this to a later revision of the sensor. 

The cheapest, but still reasonably accurate sensor seems to be the [Amphenol IR12EJ](https://www.digikey.com/en/products/detail/amphenol-sgx-sensortech/IR12EJ/7102255?utm_adgroup=Gas%20Sensors&utm_source=google&utm_medium=cpc&utm_campaign=Shopping_Product_Sensors%2C%20Transducers&utm_term=&utm_content=Gas%20Sensors&gclid=CjwKCAjwx8iIBhBwEiwA2quaq2TYPGBqj4LvuiJGPBQnxomcP6qXW0qxhU-pOLNArQF16qt4mvnFKxoCYN4QAvD_BwE), but it's $179 for one, which currently exceeds the total BOM price of the full frog sensor.

closed 2021-08-11 by keenanjohnson