Ribbit-Network/ribbit-network-frog-hardware #15: Add a Barometer to the Sensor

keenanjohnson, 2021-08-02:


keenanjohnson, 2021-08-03:
I've just purchased a few of these simple barometeric pressure sensors and will test when they arrive.

https://www.adafruit.com/product/4494

keenanjohnson, 2021-08-07:
The barometer has arrived. I'm going to install for testing shortly.

[image]

lwbantoto, 2021-08-07:
Awesome! How are you planning to test the barometer?

keenanjohnson, 2021-08-07:
Right now I’m planning on taking two sensors right next to each other and
comparing the barometer compensated sensor to the non compensated one.

Hopefully there will be a sufficient noise reduction or shift in the
compensated sensor!

keenanjohnson, 2021-08-17:
I've wired the sensor and integrated in software:
https://github.com/Ribbit-Network/ribbit-network-frog-sensor/blob/912850c6732eaf6c82549f14c60ef1ddb7b91eb2/software/co2/co2.py#L101-L102

I don't see a noticeable difference in the noise of the sensor based on my testing, however, I think this will make a big difference for sensors that exist at different altitudes. I'm making the call to include it in the sensor :)

closed 2021-08-17 by keenanjohnson