Ribbit-Network/ribbit-network-frog-hardware #16: Design Better Sensor Bracket

keenanjohnson, 2021-08-02:


keenanjohnson, 2021-08-04:
I think the easiest thing would be to create some holes in the bracket and then attach the boards with some brass standoffs like this: 
[image]

eaudiffred, 2021-08-06:
Looking for help on this one still?

I like the standoff idea.  What are the board dimensions?  I can add the holes in the bracket.

Are you set on the current layout?  Depending on the hole dimensions, you can also stack the boards using these same brass spacers.  I guess overall space in the housing would come into play then too.  

The small square board and the sensor look kind of tricky to mount.  I'd have to think about that.  

This STL file just has holes for 2.5mm threaded standoffs.  
[image]
[image]
[image]

[Sensor Clip v1.zip](https://github.com/Ribbit-Network/ribbit-network-sensor/files/6948117/Sensor.Clip.v1.zip)

keenanjohnson, 2021-08-07:
Yeah, possibly there is just a better design for this bracket altogether :). I can get you some dimensions shortly!

Thanks for helping!

keenanjohnson, 2021-08-08:
Here are some dimensions. I believe all the boards have holes for M2.5 bolts.

* [Raspberry Pi Zero](https://www.raspberrypi.org/documentation/hardware/raspberrypi/mechanical/rpi_MECH_Zero_1p3.pdf)
[image]
* [Co2 Sensor](https://cdn.sparkfun.com/assets/1/a/7/9/9/Sensirion_CO2_Sensors_SCD30_Preliminary-Datasheet.pdf) - Drawing in the PDF at the end.
[image]
* [GPS](https://www.amazon.com/gp/product/B07P8YMVNT/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1) - The GPS only has one hole weirdly, so maybe I should look for one with two holes haha. 
[image]
* [Barometer]
[image]

eaudiffred, 2021-08-09:
Here is an updated draft based on the dimensions.

- The pi, barometer, and GPS sit on the front.  The sensor sits on the back.  
- I had to guess on the hole location for the sensor though.  It looks like its attached to a board that isn't show in the PDF for some reason. 
- The GPS attachment (idk what that part is called) doesn't have mount points, so how about clipping it in place?  I've tried this before with other 3D printed parts and have had some success and some failures.  
- I also made the entire footprint bigger, based on the internal dimensions of that housing.  

Let me know what you think.

[image]
[image]
[image]
[image]
[image]

keenanjohnson, 2021-08-09:
Wow @eaudiffred this is incredible! Thanks! Great idea for the clip! Do you have an stl file or a link to the cad source so I can try to print a test version?

keenanjohnson, 2021-08-09:
And yes there is a carrier board for the scd30 sensor that I should have linked to instead of the sensor board. Here are the dimensions for that :)

[image]

https://cdn-learn.adafruit.com/downloads/pdf/adafruit-scd30.pdf?timestamp=1628525130

eaudiffred, 2021-08-09:
Do you have a measurement for the GPS attachment?  I guessed at that dimension too.  I'll update those and get you a file.

keenanjohnson, 2021-08-09:
The GPS antennae (Tan block in the picture) has these dimensions: 

L: 15 mm
W: 15 mm
H: 6 mm

You rock!

eaudiffred, 2021-08-09:
Glad you like it!

The dimensions are exact right now, so after we see how the first print comes out I can adjust tolerances as needed.  

I'm a little worried about the attachment clip being too small to get a good print, or not quite flexible enough to snap into place, but I guess we will see what happens.  

[image]
[image]
[Sensor Clip.zip](https://github.com/Ribbit-Network/ribbit-network-frog-sensor/files/6956688/Sensor.Clip.zip)

Let me know if that zip attachment opens.  I couldn't get the STL file to attach on here.

keenanjohnson, 2021-08-09:
The stl file opened just fine for me! Starting a test print now. I had to split the file into two parts that I'll glue together  because I have a tiny printer for now. A bigger printer is coming later this week luckily.

Thanks again for all your help Eric! What cad tool are you using?

https://user-images.githubusercontent.com/2559382/128767674-1a27d177-6c4f-403e-b898-7f4aa5d9e9df.MOV

eaudiffred, 2021-08-09:
Nice!  What's the bed size on the new one you have coming?

Right now I'm using Fusion 360 (you can get a free trial for one year).  I prefer SolidWorks, but it's pricy.  

This project is really a cool idea, and I love doing CAD design, so I'm happy to help!

keenanjohnson, 2021-08-10:
I got an [Ender-3 V2 3D Printer](https://www.creality3dofficial.com/products/ender-3-v2-3d-printer) which can print 220 x 220 x 250mm. :)

keenanjohnson, 2021-08-10:
Excellent news! I printed the first bracket and glued the two pieces together for a test assembly this morning! Overall this is excellent! The brass inserts I had worked perfectly in the holes provided and all the holes lined up correctly! Great work!

### Brass Inserts
[image]

### Finished Assembly
[image]

### Issues Encountered

I just ran into a few snags while putting it together that I think can be easily fixed :)

* The clip didn't actually fit in the slot at it's 100% of printing. I printed a 97% scale clip and that easily fit
* The GPS antennae that goes in the clip is actually a little taller than I thought. There is a little bump on the bottom that makes it about 8 mm tall instead of six, so the clip height will need to be increased for that
* As printed the wires had to wrap around the side of the bracket and were sorta rubbing on the enclosure as it was inserted in (see pictures below). Since the GPS clip didn't work, I was able to just route the cables through the GPS clip hole and that worked well. So we should probably add another hole for a cable pass-through. Then I just hot glued the clip on and it made a nice cable retainer actually haha
[image]
[image]
* You'll also notice that the USB cable for the GPS sticks out quite a bit in the design orientation (see picture above) and definitely bonks the enclosure going in. I was actually able to just rotate it under the barometer above, see below and that worked out fine.
* [image]
* Finally, once everything is stacked up, there isn't much width clearance for the whole stack. I definitely knocked the sensor pretty hard getting it into the enclosure. I think this would be fixed if we just made the bracket less thick. Maybe down to like 5 mm or so would be fine
[image]
The widest point using 6mm long brass standoffs seems to be about 50 mm right now
[image]

eaudiffred, 2021-08-10:
Great feedback.

What do you think about this modified layout?  More cable conscious.  

Also shrink the body width by 5mm.

[image]

keenanjohnson, 2021-08-10:
I like it!

eaudiffred, 2021-08-10:
Here are the new files.  
The thickness is now 5mm.  I adjusted the dimensions of the clip too.  I think it will work without scaling.  

[Sensor Clip v24.zip](https://github.com/Ribbit-Network/ribbit-network-frog-sensor/files/6964865/Sensor.Clip.v24.zip)

keenanjohnson, 2021-08-10:
You are the best! I'll start this printing now and report back when it's
done!

keenanjohnson, 2021-08-17:
@eaudiffred great news! In the past few days, I've successfully set up the new printer and I've successfully printed a few of the brackets above. Great work!

However, I've run into a supply chain issue where I could actually procure more than just a few of the Raspberry Pi Zero we were using in the design above. Given that, I made the decision to switch to a different board and I've updated the BOM accordingly [here](https://docs.google.com/spreadsheets/d/1lsWCXFawsJNGc44bqFXWkpjHfIFZQEj5UceJV4AiuK0/edit#gid=0)

Unfortunately, that board is quite a bit bigger and the bracket needed to be redesigned. I was impatient and you're doing more interesting work on the enclosure itself, so I went ahead and did the modeling.

I used OnShape, since it's free for public files and therefore lends itself well to Open Source like this, though maybe it's not really the best long term CAD tool. [The OnShape design is here for everyone to check out](https://cad.onshape.com/documents/0346e0daf22b10d5611dbc95/w/a8aaa87c5f1f082117b56412/e/c0f8f27ffabd2dd2476885f1).

I made a few other changes to our design and I feel good about this. I'm going to assemble one tomorrow and then close out this task if there aren't any issues found. Check out the CAD images below

Changes:
- Updated the main compute to the Raspberry Pi 4 CM + a Waveshare baseboard
- Added an antennae for the raspberry pi to increase the wifi connection range ( This will really help based on my testing)
- Increased the thickness to 7 mm so it clicks into the enclosure more securely
- Added a handle to the bottom of the bracket which makes it way easier to pull out of the enclosure locking mechanism
- I added an additional cable passthrough given the relocation of the modules

## Front
[image]

## Back
[image]

## Side Profile
[image]

eaudiffred, 2021-08-18:
Glad you got the printer up and running!  The changes look great, everything fits on there nicely.  Great idea modeling the electrical components too!  Post some pictures when you have it printed and assembled.  

I'll check out OnShape.

keenanjohnson, 2021-08-19:
Awesome! I built one today after a few revisions yesterday and I'll post pictures shortly.

Let me know what you think about OnShape. You're more of the expert on which tool is correct than me, I just know the basics of OnShape and I like that it's free to use for OpenSource, but if it's the wrong tool for the job (vs Fusion 360 ) we can definitely switch :)

eaudiffred, 2021-08-19:
I like On Shape.  Its really nice for being free software.  I've only made that housing so far, but its user friendly, intuitive, and has a multitude of functionality that I think a lot of the free stuff often lacks.  I'm on board with using it.  The file and navigation structure is great too once you get used to it.

keenanjohnson, 2021-08-19:
This bracket came together super well! I might make a few minor adjustments for cable routing but then I'm going to call this done I think for this version. 

[image]

[image]

eaudiffred, 2021-08-19:
Looks great!  Efficient use of space.  

Is the GPS antennae glued into that hole/does the clip work?  If it's not working I'll try to think of a different mounting method.

keenanjohnson, 2021-08-20:
Haha oh yeah good reminder. I actually just stuck it in there temporarily while the clip was printing but I should pull it out of there and try the clip. Orienting it in that orientation is better for the GPS antennae reception, [so in my current CAD](https://cad.onshape.com/documents/0346e0daf22b10d5611dbc95/w/a8aaa87c5f1f082117b56412/e/c0f8f27ffabd2dd2476885f1), I changed the clip dimensions to still use the clip, but keep it in that axis. I'll report back if that works haha.

keenanjohnson, 2021-08-20:
Here's the bracket with the clip installed. It seems to work well. 

[image]

[image]

[image]

I noticed that if the gps antennae cable is touching the usb cable, the gps will fail to get a gps lock indicating that the usb cable is fairly RF noisy. I probably need to add some little feature to retain the cables apart from each other.

keenanjohnson, 2021-08-21:
Ok I've made a few minor tweaks to this original bracket and I'm printing one to test now. The biggest change is I've also added some sidewalls to help protect the sensors from getting bumped against the interior ribs of the enclosure when you take it in and out. I also made some slight adjustments in placing of things to help with cable routing.

[image]

[Onshape File](https://cad.onshape.com/documents/0346e0daf22b10d5611dbc95/w/a8aaa87c5f1f082117b56412/e/c0f8f27ffabd2dd2476885f1)

I released this as V1 of the bracket in OnShape.

eaudiffred, 2021-08-23:
I like it.  Have you had a chance to print it?

keenanjohnson, 2021-08-24:
Indeed! I assembled the first one today and it looks great! [image]

[image]

keenanjohnson, 2021-08-24:
This looks good to me, so I'm going to close this ticket out as complete for this version of the sensor.

closed 2021-08-24 by keenanjohnson