Ribbit-Network/ribbit-network-frog-hardware #17: SCD30 CO2 Sensor High Humidity Bug

keenanjohnson, 2021-08-07:
Something odd is going on with one of my test CO2 sensors. About 2 days ago, it started reporting a relative humidity that was radically high and hasn't yet returned to normal. The blue line below is higher than real humidity:

[image]

I pulled the sensor out and let it dry out inside for a few hours, but that didn't change anything.

Then I tried to bake it at around 80 C for a few hours in the oven. That did seem to have improved the humidity sensing, but it seems like I fried something in the CO2 LaserElectric torchFireRightwards arrowFrowning face. 1.2K PPM is NOT correct.

[image]

SCD30 Serial Number: 2862279

keenanjohnson, 2021-08-07:
I'm not too concerned about the sensor failing after the oven testing, I must have just exceeded some thermal limit, but the humidity sensor reading incorrectly seemingly randomly seems concerning.

I'm going to reach out to [Sensirion](https://www.sensirion.com/en/) to see if they have any insight on why that would happen.

keenanjohnson, 2021-08-07:
@dahlwinters have you ever seen anything like this?

keenanjohnson, 2021-08-07:
Email out to the Sensirion: 
[image]

dahlwinters, 2021-08-08:
Haven't seen anything like this with the relative humidity. If it was with the CO2 readings only, it might have been an auto-calibration problem. Definitely something is wrong since a value of RH above 100% is not possible.

keenanjohnson, 2021-08-08:
Thanks! I'll update this issue as I learn more!

keenanjohnson, 2021-08-13:
More info from Sensirion, but no root cause.

"Thank you for the addition temperature plot. The graph shows a steady temperature drop of around 10°C and at the same time an unreal humidity increase over 100% RH. The temperature decrease is nearly linear and looks basically the same as if someone entered a temperature offset. The SHT31 on the SCD30 PCB is limited to an output of 100% RH. The only way to get values over 100% RH is setting a temperature offset as the compensation engine output is not limited to 100% RH. The average temperature offset due to self heating of the SCD30 PCB is 3°C. I’d suggest to set the offset to 3°C and see what happens.

There are a few other points to consider.

- Power supply needs to be able to provide 75mA
- There should be no pull up resister on the SDA and SCL
- Bus voltage is 3V, maximum 3.3V
- Supply voltage must be between 3.3V and 5.5V
- For code verification please check our GitHub repository @ GitHub - Sensirion/info: Overview over the Sensirion Github repositories & available sensor drivers

keenanjohnson, 2021-08-17:
Closing this issue out for now. I doubt I'll root cause the sensor issue without more help from the vendor, but I've documented what I know here for the future in case we see this issue again.

closed 2021-08-17 by keenanjohnson