Ribbit-Network/ribbit-network-frog-hardware #18: Figure out best way to paint the enclosure

keenanjohnson, 2021-08-08:
I would really like the sensor to be green-colored, since it is a frog after all.

[The enclosure ](https://www.amazon.com/gp/product/B01M64ISDE/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1) is made of an unknown plastic.

Spray paint would probably be the easiest, but given that this is a greenhouse gas project, I'd like to avoid the aerosols and the nasty volatiles contained within.

[image]

keenanjohnson, 2021-08-08:
Getting a paint sprayer like this might be the best short term option: https://www.amazon.com/Rrtizan-Electric-Patterns-Painting-Cabinets/dp/B08G81CTB5/ref=asc_df_B08G81CTB5/?tag=hyprod-20&linkCode=df0&hvadid=459681103177&hvpos=&hvnetw=g&hvrand=5572313783352482589&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9033311&hvtargid=pla-984950491839&psc=1

eaudiffred, 2021-08-10:
I reached out to AcuRite (The manufacture listed on Amazon), they provided the below response:

_The Solar Radiation Shield is constructed of **high-grade ABS plastic**. 
We do not recommend painting the Solar Radiation Shield as this may affect the accuracy of your sensor within it.
The Solar Radiation Shield is only available in white._

Not exactly sure why painting it would cause accuracy issues?  Just need to make sure its completely dry before installing the sensors.  

Also, interestingly, this thing is significantly cheaper on their website than on amazon.  Same model number.
https://www.acurite.com/shop-all/weather-instruments/weather-sensors-and-parts/accessories/solar-radiation-shield.html
$10.99 vs. $18.56.

eaudiffred, 2021-08-10:
A quick Google search for "How to paint High-grade ABS" yielded the following recommendations:

- Clean the part with mild dish soap
- Rub the surface with a very fine grade sand paper to promote adhesion
- Prime first
- then paint

What steps/paint did you try for the first one?

eaudiffred, 2021-08-10:
After a little more reading it seems like a solvent-based paint or an acrylic paint are the best options.

keenanjohnson, 2021-08-10:
Oh nice! Thanks for that! Good catch on the cheaper price :)

I would guess that they are concerned about the temperature rise from a dark color if it's absorbing too much thermal radiation? I wouldn't think this would be too significant with a light green though. Might be pretty bad with a super dark black in contrast.

Previously I used a crappy acrylic art paint I had lying around and didn't sand or prime it at all haha.

Given the amount of work it might be to paint reliably, in your mechanical engineering opinion, do you think it might just be easier to 3D print the whole enclosure using a green PETG? PETG is supposed to be super UV resistant, so it should fare ok in the sun and then we would control the whole geometry without being beholden to a vendor hah.

The downsides are that 1) A lot more CAD work would have to be done to design the enclosure and 2) It might take a while to print each one, but I'm not actually sure it's more time than the whole painting process.

What do you think?

eaudiffred, 2021-08-11:
Good call on the thermal radiation.  That's got to be their concern.  

I was thinking the same thing about the possibility of 3D printing.  Painting is a dirty process, and can defiantly be a huge pain when trying to mass produce.  

That being said, it's a tricky structure to replicate with a 3D printer, lol.  I think the AcuRite design concept is worth sticking too since it's designed for outdoor sensor shielding purpose.  I'm thinking that we could design some flat panels that assemble to create a similar functioning structure.  I have a couple ideas that may work, or at least get the ball rolling.  I'll sketch them up.  

Do any of the electrical components need to be protected from the elements?

keenanjohnson, 2021-08-11:
Yeah I think the general design is correct as well! People seem to have printed similar structures before ([see here](https://www.thingiverse.com/thing:2330852)), so it seems doable?

The flat panel idea could also be cool, but definitely harder to print.[I saw this one a while ago](https://www.thingiverse.com/thing:2936265), but I'm not sure how well it works.

The electrical components should be kept dry as in not rained or snowed on and the sensor is sensitive to direct sunlight, but a similar level of protection as the current AcuRite design has should be fine electrically. Honestly, I'm not sure how much to worry about things like spiders and stuff getting up in there, so perhaps there would be advantages to sealing the electronics more in that sense?

eaudiffred, 2021-08-11:
That first link posted seems like a great design!
What kind of angle can the printer handle without the structure collapsing during printing?
[image]

keenanjohnson, 2021-08-11:
Good question! I know the traditional rule of thumb is usually 45 degrees as a relatively conservative number. I've heard that sometimes you can push it up to maybe 60 degrees at the max, if you spend a lot of time tuning the printer and specific machine parameters. So probably the answer is something between 45 and 60 is the max, but we can always test to see!

eaudiffred, 2021-08-11:
I drew this up based on that design you linked.  Each body fin is identical and they bolt together with [M4x40mm](https://www.mcmaster.com/91292A132/) screws.  You can stack as many in a column as needed.  I just threw a lid on top for now.  It needs some work as there is currently no attachment for the electronics board, or any external hanging method.  

The sides of the fin are at a 60deg angle.  Hopefully that prints ok?
The files are attached.  I'm worried about the cost though.  Each fin is estimated at 193g (100% infill) when made with PETG.  Do you have a budget in mind for this component?  Perhaps there is a way to optimize the shape to reduce the material and still fit in the electronics.  Let me know what you think.  
[image]
[image]
[image]
[image]
[Shield Lid v12.zip](https://github.com/Ribbit-Network/ribbit-network-frog-sensor/files/6970921/Shield.Lid.v12.zip)

keenanjohnson, 2021-08-11:
Wow that looks awesome! You are making so much progress!

I don't really have a cost target right now, but just trying to think of things in terms of comparables. Right now the comparison is this 3D printed approach vs the Acuright off-the-shelf enclosure. PETG is between $20-$25 a kg. So we're probably looking at over a full roll for each enclosure with 5 shelves, so that is a significant increase from the $10.99 Acuright enclosure.

Perhaps a good compromise would be to only put some of the electronics in a vented section and the rest in a more normal closed box. Only the Co2 sensor and the barometer need to be exposed to air, but not the GPS or raspberry pi. Maybe that would significantly cut down our PETG usage?

Something like this very bad sketch I just made. I drew the vented part on the top, but it might actually make sense to have it on the side instead? Sorry for my poor sketches

[image]

or side mounted?

[image]

Also I've been told that I really should try to incorporate the eyes and a little cute mouth to make it look like a cute frog if possible haha. Not sure how that might factor into the design, but just bringing it up while we're brainstorming here to lay out all the parameters.

[image]

Perhaps the vented portion on top could be made to look something like a sitting frog with the eyes and mouth on top and a more rounded bottom haha?

[image]

eaudiffred, 2021-08-12:
I'm realizing that 5mm is probably too thick for those fin profiles.  I bet we could get away with 2 or 2.5mm.  That change, along with slimming up the mounting posts would save material.  

The sketches help a ton!  I'm very visual, so quick sketches are sometimes the best way to communicate, haha.  I like the idea of splitting the electrical components up into two groups.  I think that will allow the whole thing to be smaller too.  I might take a stab at each sketch so we can work out the pros and cons.

The frog head is difficult, but 100% worth it.  My surface skills in CAD are pretty bad, but I do have one idea that may work out.  I'll try it, hopefully it won't look like a 5 year old sculpted it.  How about something like this where the head comes up off the lid of the shield:
[image]
(That's just a google image crop.  I'll be stoked if mine actually looks that good)

keenanjohnson, 2021-08-17:
Hey @eaudiffred ! I updated the title of this issue to be "Consider Fully 3D Printed Enclosure", since that's what we've sort of arrived at.

I think we've done a lot of good work in #16 to create a bracket for the Accuweather enclosure. However, I think that designing our own enclosure would still be the better longer-term idea.

So I would propose we continue design efforts on a fully 3D printed enclosure here and I'll build sensors using the bracket in #16 for the first version of the sensor.

keenanjohnson, 2021-08-17:
I also created a new project for any other tasks that I anticipate will arise as we go down the 3D printed enclosure route: https://github.com/Ribbit-Network/ribbit-network-frog-sensor/projects/4

eaudiffred, 2021-08-18:
Sounds like a good plan to me.  I have a 1st draft almost finished.  I'll post some pictures tomorrow morning.  I'll also try and get the files into OnShape and just start using that as the CAD software.

eaudiffred, 2021-08-18:
Here is the draft.  Still needs a lid with some kind of Frog.  (that turned out to be way more difficult than I thought).  I'll move these into OnSketch.  Might take me a day or two because I'll have to redraw them (in order to get the edit/function tree).  Should be easy to share and collaborate after that though.  

I'll also make sure the PI4 fits in that enclosure.  The antenna is water proof, right?  Any thoughts on the whole design?

[image]
[image]
[image]

eaudiffred, 2021-08-18:
Some dimensions for scale.
[image]

eaudiffred, 2021-08-18:
Progress so far.
[3D Print Housing - On Shape](https://cad.onshape.com/documents/ad0cd8a667a26470e23bc74f/w/16be44f91ae0faf2beb72d40/e/1992253027dd0a9f2b426b14)

keenanjohnson, 2021-08-19:
Haha no worries on the frog. We can always just stick eyes on the top and that's great :)

The antennae is waterproof yeah :) Wow this is looking awesome! I like how you stacked the barometer on top of the CO2 Sensor! Genius!

What are you thinking for attachment points for the whole unit? Mounting features on the bottom? 

Also would we add a hole in the enclosure lid to route the wires from the box to the sensors and antennae?

eaudiffred, 2021-08-19:
Good questions.  Here are the details that are still up in the air:

- Antennae placement and positioning
- Cable routing; power coming in and then cables between pieces.  Do you have a wire diagram?  Or a quick hand sketch is fine too.  I just don't know what connects where and with what type of cable
- Cable entry and exit.  If we are worried about water, the best was to seal the enclosure is by using cord grips.  If we aren't too worried, putting the holes on the side would be best and probably keep out most water.  
- Placement of the GPS antennae
- Position of the Pi4 board.  I think its a little too cramped in there right now to get the power cord plugged in.
- Housing mounting/hanging method.  I liked the nail hook on that Acurite housing.  We can have multiple mounting options too.  

I also still need to determine the material usage and if it's in the cost ballpark we are shooting for.  
Oh, and the fins are still in question as to if they will print ok.  Its a steep angle, but I just don't have any experience printing something like that to know if it'll work or not.

keenanjohnson, 2021-08-19:
Makes sense! 

I do have a rough wiring diagram here: https://github.com/Ribbit-Network/ribbit-network-frog-sensor/tree/main/hardware, but it's now out of date after the switch in computer boards and I forgot to add the power cable haha. I'll update that shortly, but we should only have three cables:
* Power coming into the unit (this should probably come in on the bottom I would think so it's mostly shielded from rain etc
* The antennae cable and the sensor cable for the two sensors ( These would ideally go out of the top of the housing so they live in the enclosure. 

I like having multiple options as well! We could have the hook and perhaps mounting features on the bottom! 

Finally, I just checked the material properties for one of your fins via the 3D printer slicer and each fin should be able 76 g. The box and lid are 93 grams. So we're looking at probably around 250-300 grams for each enclosure I would bet, so probably 3-4 enclosures per roll of filament. Each roll is ~$24, so that's around $8 enclosure, which is definitely cheaper than our current option! So great win!

keenanjohnson, 2021-08-19:
Overnight tonight I'll try to print a fin in PETG to prove that it is possible :)

keenanjohnson, 2021-08-19:
Printing now. Let's see how it turns out![image]

keenanjohnson, 2021-08-19:
@eaudiffred Great news! The enclosure piece printed out perfectly! It looks awesome!

[image]
 
[image]

[image]

[image]

[image]

eaudiffred, 2021-08-19:
Wonderful!  How's the rigidity?  It's only 2.5mm thick.  I think that's enough for something that's just going to be hanging stationary, but hard to tell until you hold it in hand.

keenanjohnson, 2021-08-20:
It feels plenty stiff to me. I can definitely bend the middle if I really squeeze but it doesn't feel close to cracking or breaking (see video). A nice solid feel. The rounded corners feel really strong and I can't really bend them. 

https://user-images.githubusercontent.com/2559382/130164162-218ba944-4b06-43bf-8415-245723f9df55.MOV

eaudiffred, 2021-08-20:
Great!  That seems durable!

eaudiffred, 2021-08-23:
Here's what I have so far.  The 3D printed parts add up to 302g.  I think it's similar in size to the Acurite housing.  It would be nice to make it smaller though.  I want to look at the configuration a little more.  Is the ultimate goal to use the Pi-0?  

Your prototypes will also tell us how necessary it is to protect the PI from the elements.  If the Acurite housing protects it enough, this design could be changed to something more like the Acurite, but designed to fit only these components.  
[File](https://cad.onshape.com/documents/ad0cd8a667a26470e23bc74f/w/16be44f91ae0faf2beb72d40/e/7055af9693b324f45cfdcca8)

[image]
[image]

keenanjohnson, 2021-08-23:
Wow this is looking awesome! Although I would like to use the Pi Zero, I can't figure out how to buy them at a reasonable price in more than 1-2 quantities, which isn't great for the longer term of this project. So let's stick with the larger board in the cad for now.

I like having the raspberry pi in a box for some reason, but I'm not actually sure if that's better than the accuright style to be honest other than maybe less spiders or something will get in there? If we leave the box exposed like that, we'll probably need to seal the screw holes or something to prevent rain from coming in?

keenanjohnson, 2021-08-23:
Oh and we'll also need a hole in the bottom for the power cable like so: 

[image]

keenanjohnson, 2021-08-23:
FYI I've also made good progress on modeling a new Frog Head top that could be fitted onto the existing Accuright enclosure: 
[image]

[image]

[image]

[OnShape Doc Here](https://cad.onshape.com/documents/6963af519f9205453284bd94/w/8b58fe5fc28450f6ffbb9c40/e/3c52995ec56d10a3f5248627) We could either use this or adopt it to work with your new concept!

I was able to prove that I could print this geometry in PETG last night no problem!

keenanjohnson, 2021-08-23:
[image]

[image]

eaudiffred, 2021-08-23:
Having the Pi protected seems like a good idea, but idk for sure.  I'll just keep it enclosed for now.  Good call on the power cord.  I forgot the ones for the sensors too.  

That Frog Head looks great!  I'm impressed at the printer too.  I wouldn't have guessed it could support itself like that, but the print looks good.  If you don't mind, I'll copy that file and get it to fit on the new concept so you have some options.  It looks good on top of the Acurite housing too though.  Fits right in place.

keenanjohnson, 2021-08-23:
Yeah I was a bit surprised as well haha. The eyes and mouth will get printed separately in different colors of PETG and then can just get glued into their slots.

Yeah go for it!

keenanjohnson, 2021-08-23:
I like that our two design directions are converging on something similar. That makes me feel like we're on the right track. :)

daveb377, 2021-08-25:
> 
> 
> Wow this is looking awesome! Although I would like to use the Pi Zero, I can't figure out how to buy them at a reasonable price in more than 1-2 quantities, which isn't great for the longer term of this project. So let's stick with the larger board in the cad for now.
> 
Not sure what the computational requirements are on each frog are, but there are smaller processors that support SPI/I2C as well as WiFi. The ESP-32 is the first thing that comes to mind, but there are others as well.

keenanjohnson, 2021-08-25:
Great point @daveb377! A more detailed discussion on which processing option is best should be had for future versions of the sensor. I've created #16 to discuss exactly that.

eaudiffred, 2021-08-31:
- I guess it kind of depends on how protected the inside needs to be.  If we are shooting for water tight, then we can put a gasket around the edge of the lid to seal it against the box.  We would also need to use cord grips to seal the holes in the box where the cables pass through.  I'll work on a rendition of that so you can see.  
- I'll see if I can fit it on the top with a cord grip.  
- See the pictures below.  Can the printer print this structure?  I was worried about the bridge failing.

[image]
Could also do something like this:
[image]
Or thin vertical slots.

keenanjohnson, 2021-09-02:
Yeah I think we'll have to test possibly in both cases. Perhaps a full seal isn't required, but just want to avoid actual water flowing inside the enclosure if rain hits it, but perhaps that won't be an issue?

Same with the slots, we'll need to test I think. The triangles generally pretty better I believe.

eaudiffred, 2021-09-13:
I added an [O ring](https://www.mcmaster.com/1209N1/) and [cord grip](https://www.mcmaster.com/69915K47/), to the [assembly](https://cad.onshape.com/documents/ad0cd8a667a26470e23bc74f/w/16be44f91ae0faf2beb72d40/e/7055af9693b324f45cfdcca8).  

The only other problem area is the port on the side.  To get it water tight, it'll need a cord grip.  I would try and refrain from putting it on the top, that will allow water to get in easier than on the side.  

I'm thinking this is overkill though.  With the inset lid and the cable holes on the bottom and side I don't think you'll have water issues.  Moisture/humidity would get in, but I don't see rain water actually getting in there and pooling or dripping.  

[image]
[image]
[image]

keenanjohnson, 2021-09-15:
Wow that looks great! Yeah what if we just moved the slot on the side to the top so that it's fully protected by the top shield?

Do you think it's ready for me to do a test print of things?

eaudiffred, 2021-09-20:
I think it's ready now.  The cable hole is moved to the top and I added a nail mount to the back.  
[Assembly ](https://cad.onshape.com/documents/ad0cd8a667a26470e23bc74f/w/16be44f91ae0faf2beb72d40/e/7055af9693b324f45cfdcca8)

Here is the parts list: (These can be purchased from anywhere, just have to be the same size/dimension.)  
[O Ring](https://www.mcmaster.com/1209N1/)
[Cord Grip](https://www.mcmaster.com/69915K47/)
[M4 x 0.7mm x 30mm](https://www.mcmaster.com/91292A130/)
[M4 x 0.7mm x 10mm Button Head](https://www.mcmaster.com/92095A190/)

keenanjohnson, 2021-09-21:
Awesome! I'll print those up and let you know how it comes together! You rock! This looks great!

eaudiffred, 2021-09-21:
I added a rubber grommet to the cable hole on the top of the enclosure.  Hopefully you haven't started printing anything yet.  Not a huge deal if you already started.  I just thought it would add a little more water resistance to that hole.  Depending on how small the plug is at the end of those cables I can shrink the hole even more.  This kind of thing can be a work in progress though.  Not critical for prototyping.  

[Rubber Grommet](https://www.mcmaster.com/9600K54/)

[image]
[image]
[image]

keenanjohnson, 2021-09-21:
Thanks @eaudiffred! No, I hadn't started printing yet, so I'll give those a shot!

keenanjohnson, 2021-09-27:
Ok @eaudiffred ! I printed out some of the components and discovered a few issues:

* Easy fix: The hole for the power cable is mis-aligned with the connector, so it can't mate :) (See pic below)
* Harder fix: I should have caught this when looking at the CAD, but the power connector and the antennae cable interfere which will be pretty bad since the antenna cable can break sort of easily. I'm thinking that the easiest thing might be to change the antenna so it sticks out of the bottom of the enclosure like I show in a picture below
* The GPS mount works well, but the GPS cable sticks straight up, which isn't great because that cable is pretty weak. I'm thinking the best solution would be to cut a slot in one of the clips for the cable

[image]
[image]
[image]

keenanjohnson, 2021-09-29:
[I copied your awesome start to the enclosure @eaudiffred and resolved the issues I mentioned above.](https://cad.onshape.com/documents/cd32919c9eaa1d3ab0b28c0f/w/bb64da85bf9953acf5e97e52/e/937985b68922d664cb7d9047) Printing a test piece now and I'll report back if that resolves everything! Thanks again so much!

[image]
[image]
[image]

eaudiffred, 2021-09-29:
Ahhh, sorry about that.  I should have caught those first two mistakes.  Thanks for fixing them up, looks good.  

Since the antenna is now outside, we could shrink up the enclosure to save some material and space.  Would that be beneficial?

keenanjohnson, 2021-09-29:
No worries! It's all good because the USB connector wasn't in the model previously, so I didn't see it until I did the print. You did the heavy lifting with the initial concept!

I would like to make it smaller just to minimize print time, but unfortunately, we have a USB cable coming out of the top of the board as well to connect the GPS Unit (That's also not modeled, but I should fix that haha). I'll post a picture soon with the electronics in the box and you'll see what I mean. Once I put it together and get a better sense of the bend radius for that cable, I'll have a better sense of where it can shrink I think.

keenanjohnson, 2021-09-30:
Ok here are the pictures of the build of my cad above. You can see all the space that the stupid GPS USB cable takes up haha.   I think there is room to move the GPS antenna above the USB cable and shrink the whole box horizontally, but there isn't really room to shrink vertically. I just ordered a right angle USB cable I'll try to see if that saves us any vertical space.

[image]
[image]
[image]
[image]
[image]
[image]

eaudiffred, 2021-09-30:
Those cables take up way more space than I thought!  What do you think about shifting the GPS antenna down to the bottom, and then sliding the wifi (?) antenna back up into the box along that whole right side?  Is it astatically an issue to have that antenna hang out? 

Hopefully the right angle cable buys some space.  There is no way to change the Pi layout, is there?  Move the power port to the other side with the other cables?  

The external components on top seem to fit in-between the fins.  Any issues up there?

I haven't had a chance this week to play around with it.  Hopefully I get some time this weekend.

keenanjohnson, 2021-10-07:
I added the 3D model for the GPS Unit and USB cable into the 3D model. I had to move the mount holes over slightly and adjust the opening, but everything fits just barely haha. You can see the sensor stack up I have, which I've tested physically and seems to work nicely. Let me know if this workflow of copying the assembly each time works for you. We could agree on one document and both edit it, or continue with this workflow while we're making lots of changes.

[Here's the new CAD Assembly.](https://cad.onshape.com/documents/b3e6eeabf50d585d20f25fc6/w/a4a82302d129f025f23b244a/e/634db96a0c5d4c7b0ff00c45)

[image]
[image]

eaudiffred, 2021-10-07:
Looks good.  Close fit, but still clearance.

Up to you on the work flow.  I think it would be easier if we could both work off of the same document.  However, we might loose some of the version history that we create by creating these copies.  What do you think?

keenanjohnson, 2021-10-07:
All right the print is finished and things look generally good! The fit is really snug and I had to press on it quite a bit to get it to pop in, so we might want to look at increasing the tolerances in the groove slightly I think to account for a bit more variation in the production.

Just one issue I noticed when screwing it together: because the bolts for the head and sensor shrouds before the box lid goes on, there isn't enough room to get a standard Allen Key to put in the top two bolts for the electronics box (hopefully it's clear in the pictures below). Perhaps the best solution is to move those top bolts down so they aren't in the corners, but instead in the upper-mid section below the sensor shroud so they can be easily installed (see pictures below)

Let me know what you think! As for the OnShape Assembly, yeah it could be nice to just work in one Document. That's how I've worked at companies with OnShape before. To keep the history, [we can each just create a "Version" of the document ](https://cad.onshape.com/help/Content/versionmanager.htm) each time either of us makes a change which will allow us to keep all the revision history! [I believe my assembly here is the most recent,](https://cad.onshape.com/documents/b3e6eeabf50d585d20f25fc6/w/a4a82302d129f025f23b244a/e/634db96a0c5d4c7b0ff00c45) so I'll add you as a collaborator there and let's just create the versions in that assembly going forward. Shoot me an email or text if you have any issues or need help figuring that out!

[image]
![IMG_9781](https://user-images.githubusercontent.com/2559382/
[image]
136433591-cb939d95-dfdb-4aeb-a861-5ce48ddd07e1.JPG)
[image]
[image]
[image]
[image]

eaudiffred, 2021-10-07:
You are getting these turned around quick!  I'll increase the tolerance by shrinking the tooth.  I think its only 0.05mm smaller right now.  I'll increase it to 0.2mm tolerance total, so 0.1mm on either side.

The version history is cool.  That's defiantly the way to do it.

The bolt position is a bummer.  I'm afraid that moving them lower will cause space issues with the antenna or the Pi.  The groove for the lid tooth takes up a lot of that edge space.  What do you think about widening the shroud, or modifying it somehow to allow the tool to fit?

I'll explore both options (moving the bolts and expanding the shroud) and see what I can come up with.

eaudiffred, 2021-10-18:
I added two posts instead of bolts.  I loosened the tooth/groove tolerance too.  

[image]

keenanjohnson, 2021-11-02:
As we discussed @eaudiffred let's release this as a version 2.0 of the Frog Sensor alongside the new compute (discussed in #62). I've added it to the Frog Sensor V2 Milestone.

eaudiffred, 2021-11-03:
Sounds good.  This is the board, right?
[BEAGLEBONE_BLACK.pdf](https://github.com/Ribbit-Network/ribbit-network-frog-sensor/files/7464940/BEAGLEBONE_BLACK.pdf)

Where with the USB and power cords be plugged in?  Is it just these two that I've circled?
[image]

eaudiffred, 2021-11-03:
The boards are very close in over all size.  The bolt holes will have to be shifted a little, but it won't really make a huge difference.  The biggest change will be the location of the plugs/cords and the larger wifi dongle.

keenanjohnson, 2021-11-03:
Yup! The new power jack is going to be the barrel plug on the right.

[31 more replies]

closed 2021-12-01 by keenanjohnson