Ribbit-Network/ribbit-network-frog-hardware #22: Design Raspberry Pi "Hat" Printed Circuit Board (PCB) for Sensors

keenanjohnson, 2021-08-19:
A big way to drive the cost of the Frog down is to design a custom circuit board that contains some of the electronics, instead of buying multiple dev boards. We should explore this option :)

keenanjohnson, 2021-08-19:
@lwbantoto has been taking the lead on this, so assigning this task to him :)

daveb377, 2021-08-22:
Is this a Pi-Had or a PCB for the compute module?

daveb377, 2021-09-03:
SO a Pi Hat with a Sensiron sensor and a barometer? What else needs to be on there? Battery Interface/Solar power panel hookup?

keenanjohnson, 2021-09-04:
Probably eventually a battery / solar hookup, but we'll need to figure out how to reduce the power consumption from the processor before that makes sense due to the lack of a sleep mode in the Raspberry Pi.

Yes the session sensor, barometer, and a GPS sensor are the intent for this hat.

daveb377, 2021-09-08:
I spent 20 minutes looking for a Pi hat with those sensors with no success. I'll take a look at a board design, it doesn't look too difficult, although I might have some Pi related questions. First I need to out a Kicad problem with another project, it might be a week or two.

keenanjohnson, 2021-09-08:
Sounds good! Reach out with any questions!

daveb377, 2021-09-23:
@keenanjohnson  - What sort of requirements are there on the CO2 sensor? The Sensirioin part is fairly large and expensive, and I'm wondering if there's a smaller part that would work? I'm trying to fit everything onto a Pi uHat for the Pi Zero, and it looks like it will all fit.

keenanjohnson, 2021-09-23:
Hey @daveb377 we don't have any hard requirements on the Sensiroin sensor persay as we're still trying to identify the exact requirements from the data scientists. However, here are the loose requirement set:

* Humidity and temperature measurement ( The sensirion part includes an on board temperature and humidity sensor which is really important for the data analysis). 
* A CO2 PPM accuracy of +/-30 PPM
* Good temperature stability
* NDIR sensing technology

If there is another part that fufils these requirements at a lower cost / size, I'm definitely open to it, but haven't found it yet. 

It's great that you are helping to tackle this, but here's a few other key pieces of information for you that might make the Hat concept obsolete.
1 - We've switched in the BOM to the Raspberry Pi 4 CM4 module which includes a much higher gain antenna given I had some wifi connection issues with the small on PCB antennae on the Raspberry Pi Zero. So if we stick with the Hat format, we should probably size it for a normal raspberry pi.
2 -[ In another task](https://github.com/Ribbit-Network/ribbit-network-frog-sensor/issues/18), we've been working on a new enclosure concept to 3D print the full enclosure. Right now our concept has trended towards placing the sensors in a different enclosure section than the Pi, so I'm not if the hat concept really makes sense anymore? I'm open to people's opinions on this :). It would certainly make the whole package smaller if 
[image]

lwbantoto, 2021-09-23:
@keenanjohnson Are we satisfied with the humidity and temperature measurements from the Sensirioin part? Perhaps we should add our own temp sensors (thermistor) and humidity sensors on-board.

keenanjohnson, 2021-09-24:
They seem reasonably good and I've been told that Sensirion makes the best in class humidity sensors. I'm open to other options though if we feel there is a real advantage to another sensor combination.

lwbantoto, 2021-09-24:
For our application, redundant sensors is unnecessary right?

keenanjohnson, 2021-09-24:
If it's free or very low cost it doesn't hurt I think, but I definitely
don't think it's necessary here. For temperature we already have two
temperature readings: from the barometer and from the SCD30.

daveb377, 2021-09-24:
From a $$ point of view, the Sensirion is fairly expensive and large, $67.27/each at Digikey, so it's pricy compared to the Sensirion SCD41, which is $37.10/each at Mouser. I'll compare the specs for the two. 

I'll also price out the various Raspberry Pi options. From an assembly point of view, fewer wires and interconnects to make is much easier to assemble and build, and causes fewer issues down the road, so it's for sure a better option if you plan to build more than 10 or 20 of these.

keenanjohnson, 2021-09-24:
Yeah I definitely agree that a hat form factor would be nice for assembly :).

Looking forward to your comparison! I do still think the hat form factor makes sense :)

daveb377, 2021-09-30:
Pulling together a comparison spreadsheet in Google Sheets here :  https://docs.google.com/spreadsheets/d/1TGryxRZgJ1XtoPHgSFU0XBkzrDwEvxxF2vDguaEpcZk/edit?usp=sharing

There's 4 raspberry pi's costed out, i need to work on the capabilities of each of them. I also threw in an ESP32 Arduino Feather. It's a WIP, but I've opened it up to comments. 
Costing is pretty important, generally a multiplier of 2.5 to 3 to go from the BOM (Bill of Materials) so shaving Dollars here makes the final system cheaper by a 3:1 ration.

keenanjohnson, 2021-09-30:
Wow this is an awesome start @daveb377 and related to another issue: #31.

One other important dimension in your analysis to consider is the ease of purchasing. Right now the electronics shortage is making it really difficult to procure certain things. For example, I haven't been able to find Raspberry Pi 4's at any of the major suppliers at normal prices recently. I'm told by vendors that the shortage will probably continue well into mid 2022, so it doesn't seem like this issue will be mitigated in the short term.

daveb377, 2021-10-01:
No, I think every place I looked had had them out of stock. The other thing I was going to add is CO2 sensor options. I've poked at the Sensirion parts, but i'm not sure what else is out there

eren-rudy, 2021-10-03:
Have a few comments/questions:

### CO2 Sensors
I created this document a while ago to track CO2 sensor research: https://docs.google.com/spreadsheets/d/1RpaTCX8PqUjehLUbW_36fwdgL5wZA7mf5JP_UX7CvC8/edit#gid=0.

I've just added info for the SCD40 and SCD41. Similar to @keenanjohnson, I've done a bit of looking around for other options but there doesn't seem to be too much out there that is competitive with the price of the Sensiron sensors. @daveb377 (or keenan), if you find other sensors and want to contribute to the spreadsheet, let me know and I can add you.

Regarding the SCD4x series sensors, they do not use NDIR, but rely instead on photoacoustics. From what I understand, it is at least similar to NDIR in that it performs a 'direct' measurement of the CO2 content of the air ([this review](https://www.ncbi.nlm.nih.gov/pmc/articles/PMC7248969/) article seems pretty interesting, the first few paragraphs hint at the similarity to NDIR).  However, it does have a slightly lower accuracy (40 ppm for the SCD41 vs the 30 ppm of the SCD3x, although the variance in accuracy may be less due to the smaller total range).

### Enclosure

@keenanjohnson, I just wanted to clarify why the new enclosure design has moved the sensors. Is it to increase the integrity of the CO2 measurements by moving it away from the heat source of the RPIs SoC?

### 'Hat' PCB form factor

@keenanjohnson, I believe you mentioned that you would like to keep the 'hat' form factor to that of the 'standard' RPI board. Wouldn't a form factor the size of the CM4 satisfy all of the same requirements? Above, you cited the WiFi connectivity as a reason to use the standard RPI form factor, but isn't the U.Fl connector on the CM4 itself? Am I misinterpreting something here? 

That being said, maybe there are other reasons to stick with the larger 'standard' form factor for the hat? Would be curious to hear thoughts on this. The only other things I was thinking are perhaps increased supply chain flexibility and more standardization for people who wanted to use their own RPI boards.

daveb377, 2021-10-04:
@eren-rudy That's a nice summary. I copied that into the compute options spreadsheet to keep everything together. 
The CM4 has a unique connector that's not compatible with the other Pi's, which have a common connector. The CM4 is expensive, and focused for high volume industrial users, so I suspect it's harder to find. It would make for a smaller sensor I think but less each to source. 

It sounds like adding a GPS receiver to the sensor would be helpful from a wiring point of view.

keenanjohnson, 2021-10-04:
@eren-rudy yes we're considering moving the sensors for a few reasons:
* Moving the sensors away from the main processor reduces any thermal coupling between the sensor and the processor
* Moving the sensors, allows us to seal part of the enclosure providing better environmental protection for the more sensitive electronics (processors)

These two reasons would seemingly lead us away from a raspberry pi hat format. However, I do see the benefits of standardizing on a form factor. I think what might be the best compromise is to stick with a standard connector like QWIIC which some of our sensors already use. 

The mechanical design is still a prototype though, and it's not certain if we "need" to decouple the compute from the sensors. What do you all think?

eren-rudy, 2021-10-05:
@daveb377 No worries, its nice to have everything consolidated. 

Yeah I think I agree that the larger hat form factor makes sense for standardization and sourcing flexibility. 

@keenanjohnson Thanks for the clarification, that makes a lot of sense to me! 

I don't really see too much strong reason for single-board consolidation, at least for now. Maybe someday, if there was high enough volume, it would be advantageous to have everything on one PCB to drive costs down (reduce wiring, connectors, and assembly complexity). For now, I think it makes sense to explore multiple avenues to see what works? I'm sure we could find ways to study the effects of those two reasons (thermal coupling and environmental protection of sensitive circuits), but perhaps that is a larger conversation that we can have elsewhere.

daveb377, 2021-10-05:
Hmm. A QWIC interface might be very interesting. Have to look into that. 
As far as the rational for a custom board, it would certainly make assembly easier, and cables are a continual problem area. Getting boards build in small quantity is surprisingly inexpensive now, Parts availability is probably more of an issue.

keenanjohnson, 2021-10-05:
Yeah I would be curious if we can achieve a cost savings vs buying the individual boards for each sensor!

daveb377, 2021-10-19:
[image]
Not sure this is legible, but here's a first cut at a dedicated Pi Hat. I've got the two sensors plus a GPS receiver with an integrated antenna. 
My next step is to see about adding a battery management/Solar cell controller, but I need to do a little research on this first.

keenanjohnson, 2021-10-29:
Hey all! Great collaboration here! Seems like a lot of the nitpicks probably aren't high value to discuss until we have a properly finished design that we can do a real design review on :).

daveb377, 2021-12-04:
So time to pivot to a Beaglebone design? No worries, it's easy to do. I wonder if we can build a WiFi node on the Beaglebone?

keenanjohnson, 2021-12-04:
At least in the short term, [we've pivoted to a Beaglebone Black design](https://github.com/Ribbit-Network/ribbit-network-frog-sensor/issues/62) until Raspberry Pis are readily purchasable (I'm told this will most likely be about 6 months from now).

[Given that and the current mechanical design](https://github.com/Ribbit-Network/ribbit-network-frog-sensor/issues/18), I'm really not sure if this is a priority because the optimal mechanical design orients the computer board in a non-optimal way for a hat format. I think what might be the best compromise is to stick with a standard connector like QWIIC which some of our sensors already use.

It's up to you whether you would like to continue work on this or not @daveb377! I don't think it's a priority, but I'm sure we can figure out a way to use it if it exists in the future!