Ribbit-Network/ribbit-network-frog-hardware #30: Corrosion protection for the electronics

keenanjohnson, 2021-08-25:
Right now the electronics are placed in an unsealed enclosure. We need to determine if we need some stronger form of corrosion protection for the electronics. This may be especially relevant if someone wants to deploy a sensor in an area near the ocean that has a high salinity content for instance.

eren-rudy, 2021-08-31:
mgchemicals has an informative introductory resource on conformal coating technologies [here](https://www.mgchemicals.com/category/conformal-coating/).

It seems that using an acrylic or silicon coating would work for our application. The acrylic is slightly cheaper, but apparently has pretty strong fumes during use (though not once cured). The other options (polyurethane and epoxy) would also be feasible, but are more impractical due to cure time and handling/mixing considerations, respectively.

These can be easily purchased on Amazon in a bottle containing a liquid that can be applied using a brush:

- [Acrylic conformal coating](https://www.amazon.com/MG-Chemicals-Premium-Acrylic-Conformal/dp/B07B8RY7M6/ref=pd_di_sccai_4/131-1889336-7631845?pd_rd_w=mAfPp&pf_rd_p=c9443270-b914-4430-a90b-72e3e7e784e0&pf_rd_r=NKTWWGXQWHZBMXXNT111&pd_rd_r=3ef8c095-7ee1-48e6-a981-17664780b6e4&pd_rd_wg=dGDJM&pd_rd_i=B07B8RY7M6&psc=1) (It might be worth noting that there were a lot of complaints in the Amazon reviews about the quality of the provided brush affecting the ability to apply the coating. This could potentially be improved by using a different brush.)

- [Silicon conformal coating](https://www.amazon.com/MG-Chemicals-Silicone-Conformal-Coating/dp/B085G42TGS/ref=pd_di_sccai_1/131-1889336-7631845?pd_rd_w=ikoLB&pf_rd_p=c9443270-b914-4430-a90b-72e3e7e784e0&pf_rd_r=C1RH2MZR7KF9B4RSFSR9&pd_rd_r=99c06e98-05c6-4efe-a271-da3617300eb3&pd_rd_wg=fOCae&pd_rd_i=B085G42TGS&psc=1)

Both of these products can also be purchased as Aerosols, but I figured for potential safety and environmental concerns, the brush-application method might be more desirable. 

@keenanjohnson it might be a good idea using one of these before sending out the pre-assembled Alpha units that you plan on assembling. I'll leave it up to you which specific product you want to use, since you will be the one using it -- not sure if you think its worth dealing with acrylic fumes!

---

In the long-term (if we are doing higher volume PCB manufacturing), it should definitely be feasible to have a conformal coating applied by a PCB assembly service.

keenanjohnson, 2021-08-31:
Ah yes, I've used those a bit to spot fix areas on boards where I rubbed the conformal coating off of products I've fixed :). Yeah, the aerosols are for sure a mess without a big chamber/application room, so I think we want to stay away from those for several reasons, including that it requires some sort of complicated masking situation to prevent the spray from getting into connectors.

Given that all of our electronics have lots of connectors/areas that can't be conformally coated (vents for the barometer and co2 sensor for example), I think the main thing is just to identify what are the high-value areas to conformally coat with brush.

@eren-rudy would this be something you could help me map out?

Seems like trying to hit most of the raspberry pi compute module top and bottom (avoiding connectors) is the highest value area. Looking around at the rest of the electronics, maybe the GPS module and some of the ICs on the bottom of the raspberry pi carrier?

[image]
[image]
[image]
[image]
[image]

eren-rudy, 2021-09-01:
I understand that we don't want to apply any conformal coating to sensors that need air exposure, however, I want to clarify a few things: (1) I'm curious about the reasoning behind not conformal coating the connectors, and (2) is there a cost-protection tradeoff that you are weighing when thinking about selectively coating high-risk areas?

RE: (1), I understand that coating the unmated connectors would make it such that they cannot be used, but, if its not inhibitive to assembly, I think there might be value to coating the mated connector.

RE: (2), my *guess*, is that -- although it may obviously be more valuable to coat a small-pitch IC -- there would still be non-trivial value to even just coating the soldermask and other traces.

Of course, I don't have any personal experience with applying conformal coating, so please let me know what you think! I'll definitely do more research into my assumption for point (2), since coating the whole board may be overkill. I'll be sure to share any useful resources I might find.

kelp, 2021-09-01:
It also seems like the GPS requirement means you have to mount it somewhere that it's not under a roof. It seems like it would simplify the BOM if the GPS was left out, and you just had to provide lat/long at setup time. Then it could have a little more protection from the elements.

Also PurpleAir outdoor sensors are just open at the bottom and they seem to do fine. They do warn you to arrange the power cable so any moisture drips off it, and doesn't run into the electronics. However, I haven't taken one apart to see what, if anything, they have coated to avoid corrosion.

keenanjohnson, 2021-09-01:
@kelp! It's an interesting idea, but for climate data, the sensor needs to be outside anyway, so it will need to be protected from the elements either way.

eren-rudy, 2021-09-03:
Hey @keenanjohnson. I appreciate the input, that all makes sense to me!

Regarding your earlier comment, I do want to clarify exactly what you meant by mapping out the areas to conformal coat. Is the intention to add to your assembly instructions by including information for users to apply the coating themselves?  

If so, let me know. I can annotate the images and put them into an additional step in the assembly instructions. 

My only concern about this is that it would increase the effective cost, since for the most part I imagine the conformal coat/silicon grease might not be used for other things after the user builds the Frog. In that case, maybe we can add some "optional but encouraged" steps to the assembly instructions? I guess it may not be a huge problem, since many people who want to build it themselves wouldn't mind having extra chemicals like that lying around. However, it does introduce a conundrum where the DIY version could be more expensive than the pre-assembled version.

Let me know what you think!

keenanjohnson, 2021-09-03:
Yeah @eren-rudy ! That's exactly what I was thinking. Just annotating some pictures to indicate where to apply the conformal coat would be super useful!

Yeah I hear you about the price concern, but I think it's a bit unavoidable that building it yourself will be a little more expensive than purchasing one. That's just how it goes if something is being produced in bulk vs a one-off. The good news is that people can decide whether they want a conformal coat or not when they build it.

eren-rudy, 2021-09-12:
Added some simple instructions in [PR 45](#45).

closed 2021-09-16 by keenanjohnson