Ribbit-Network/ribbit-network-frog-hardware #31: Consider alternative compute board

keenanjohnson, 2021-08-25:
Currently the Frog Sensor uses the Raspberry Pi CM4 which is nice because of its ease of procurement, flexibility, and ability to use the Balena update service. 

However it's definitely possible that a different compute option would be better either for power, size or cost. This issue is to capture a detailed discussion of the trade offs.

keenanjohnson, 2021-08-25:
One of the main problems with the raspberry pi is that it doesn't seem to have an easy sleep mode, meaning it consumes a lot of power when considering creating a solar powered sensor. 

I've heard Esp32's are great at this, but creating a fleet wide firmware update system is quite difficult it seems.

keenanjohnson, 2021-08-30:
It's a good suggestion and perhaps the right call. I would love to see what a proposal looks like for using a microcontroller based board like an esp-32. 

Although it is possible to setup remote firmware updates like that, doing them at scale in a secure and verifiable way definitely get's tricky to configure a VPN and everything for all the devices.

Adesanya-Toba, 2021-08-31:
With regards to fleet wide firmware updates, I believe Amazon-FreeRTOS provides a relatively simple way to do this. It works with the AWS OTA Manager Service and this should provide the tools for fleet management.
https://www.digikey.be/nl/articles/updating-firmware-at-the-edge-using-amazon-freertos

Plus, it supports the ESP32.

keenanjohnson, 2021-08-31:
@Adesanya-Toba interesting! Yeah, I didn't know about that AWS service, but it's definitely an interesting alternative. I would be very supportive of someone selecting the right components to build a prototype using AWS-FreeRTOS based on an ESP32 device! Let me know if anyone wants to help put together a proposal there! @lwbantoto @daveb377 @Adesanya-Toba

djgood, 2021-09-09:
An alternative to messing with sleep modes is to schedule a power on via a RTC alarm for some future polling period and then just power off the device. This requires some additional hardware that I don't believe the Pi has on board, but it looks like this sort of thing already exists for the Pi, which is nice: https://github.com/PiSupply/PiJuice

One thing to watch out for with this approach would be that the boot time isn't too large proportional to the polling period for the sensors connected to the device because it could get to the point where you're not really saving power with this approach compared to just leaving things on.

If you do hit that wall (which depends on the data sampling requirements) you could speed up boot times by running a more lightweight Linux image. Or you could have a "low power" subsystem running an efficient micro which holds on the last however many samples of data in memory which is polled at a much lower rate by the power hungry Linux system whenever it wakes up.

maggie0002, 2021-11-02:
What are the advantages of the compute board and module options over having the pre-made raspberry pi devices such as the pi zero and more recently the pi zero 2? I noticed you are using Sd cards anyway which is usually the down side.

keenanjohnson, 2021-11-02:
Hey @maggie0002 the main advantage of the compute board is the integrated / pre-FCC certified high gain antennae connection. Our devices go outside, so getting the maximum connectivity strength is important versus the on-pcb antennas on things like the pi-zero's. 

The challenges we face with all the Pi devices right now are:
* Raspberry Pi's seem to be impossible to buy in the US right now and I'm told things won't get better for a few months at least.
* Power consumption is quite high and there is no proper sleep mode. We really want to do a solar powered sensor and the Raspberry Pi's "always on" power consumption is really just too high for a system like that to be reasonable.

maggie0002, 2021-11-03:
> Hey @maggie0002 the main advantage of the compute board is the integrated / pre-FCC certified high gain antennae connection. Our devices go outside, so getting the maximum connectivity strength is [...]

Ahh yes, I have experienced the dilemma. Particularly the pi zero which is limited most places to one unit at a time. Not so much and issue for open source projects where people build their own, but certainly a bummer for distribution. The next best thing I played with is the orange pi zero’s. They have a much stronger wifi, small aerial much more processing power. But Balena’s support of them is somewhere between poor and nonexistent. 

I’m still yet to find the perfect balance. Those pi compute modules setups are just such a big leap in cost, hard to find the middle ground.

keenanjohnson, 2021-11-03:
Indeed. Yeah unfortunately lack of Balena support is a deal-breaker for us right now, as we need good OTA capability.

dtischler, 2021-11-03:
The Pi Zero 2 is en-route to our Devices team, to enable support.  No ETA on actual release, as we have to get them in hand, determine the level of effort, and then integrate into the build system.  But, it's definitely forthcoming.

keenanjohnson, 2021-11-03:
Yeah I'm really excited about the Pi Zero 2, but unfortunately, they are also sold out everywhere and the RiPi team can't comment on when they'll be reliably in stock anywhere.

dtischler, 2021-11-03:
Aaaaah, good point.  Yes, the real issue is then "what can be purchased in quantity"... Let me give this a think and compare to what is currently supported by balena, could be supported by balena, or the trade-offs of going down to Cortex-M class.

keenanjohnson, 2021-11-03:
Thanks @dtischler ! In #62 it seems like the next best option that's widely available is the Beaglebone Black, but open to other opinions / options there is something better!

keenanjohnson, 2021-11-03:
Hey folks! This issue has had some great discussion, but think there are two development ways forward that are more specific than this vague issue, so I'm closing this issue in favor of the two new tasks. Let's move any discussion to those relevant issues below related to Raspberry Pi backup options and solar-powered options:

#62 
#63

closed 2021-11-03 by keenanjohnson