Ribbit-Network/ribbit-network-frog-hardware #35: Review Assembly Instructions

keenanjohnson, 2021-08-28:
In #21 I created the first draft of the assembly instructions. @eren-rudy I would appreciate a review of those documents and feel free to make any additions you see fit!

I created this ticket to track that work and I'm assigning it to you.

keenanjohnson, 2021-08-28:
For some reason, I can't yet assign it to you haha. Maybe you have to comment here first to confirm.

eren-rudy, 2021-08-30:
Hey @keenanjohnson, happy to give this a review! And thanks for re-organizing the tasks as well, I think there was some overlap.

Were you thinking that this ticket should include a full assembly to test out the instructions? I want to get a sensor running anyways so I'm happy to get the parts ordered this week and put it together as soon as I can!

keenanjohnson, 2021-08-30:
I think that maybe let's do a review here of just proof reading the docs and then whenever you have parts for a build, we can creat news issue(s) for that :)

keenanjohnson, 2021-08-30:
Done! Thanks @eren-rudy !

closed 2021-08-30 by keenanjohnson