Ribbit-Network/ribbit-network-frog-hardware #39: Add an Icon

dtischler, 2021-08-30:
I was going to PR one in, but, it probably makes more sense for you to choose what you'd like.   :-)

Need to add a snippet like this:  https://github.com/balenalabs/balena-sound/blob/master/balena.yml#L27 into your balena.yml file located here:  https://github.com/Ribbit-Network/ribbit-network-frog-sensor/blob/main/software/balena.yml#L11

🐸

dtischler, 2021-08-30:
Oh, and, need to do another `balena push` after that.  🙂

keenanjohnson, 2021-08-31:
Oh thanks for pointing that out @dtischler ! Another question for you!

[What does the version number mean at the end of the balena.yml file?](https://github.com/balenalabs/balena-sound/blob/1bf2d15f9dc5adb16d9826fd0bc6fbc0ed6b47fc/balena.yml#L44) Is that a version of the balena application or of the file format or what?  I couldn't find that documented anywhere.

keenanjohnson, 2021-08-31:
After a balena push, I have confirmed that [I see the logo](https://hub.balena.io/g_keenan_johnson1/ribbit-network)! 

[image]

keenanjohnson, 2021-08-31:
Oh nevermind, I see now that it just an arbitrary software version for my app.

closed 2021-08-31 by keenanjohnson

dtischler, 2021-08-31:
Yep, you got it, just a versioning for your own purposes...and, logo looks great!

keenanjohnson, 2021-08-31:
Soon we'll have a proper logo instead of a pic :). Thanks for the help though! Balena Open Fleets are incredible! Super excited for whenever a Local UI is released as the documentation hints :)