Ribbit-Network/ribbit-network-frog-hardware #48: Painting the enclosure green can cause it to heat up

spestana, 2021-09-16:
These Stevenson screen enclosures are white so that they reflect most sunlight and avoid heating up (and therefore heating up the air in/around them). Painting them green (the peak of the solar spectrum!) is probably causing them to heat up.

I realized this when I noticed the air temperature measurement jump ~5 C as soon as the sun hit the enclosure.

[image]

spestana, 2021-09-16:
And here's another sensor heating up at about 9:10am local time, which must be when it was first exposed to direct sunlight today:
[image]

keenanjohnson, 2021-09-16:
Yeah it's a problem I've thought a little bit about and it's a real tradeoff between the branding/marketing which increases adoption and deployment of the sensor vs the heating challenges. I did some initial testing where I put a temperature sensor in a painted enclosure vs a white one and placed them in the sun. The paint definitely causes a noticeable thermal rise (2-5 degrees c greater than white, but the temperatures are still well within the stable thermal range from the sensors data sheet (0-50 C), so I didn't think it was an actual problem.

keenanjohnson, 2021-09-16:
@spestana looking at one of the sensors from this morning, it looks like yours actually went offline around the peak temperature. Is that because you powered off the sensor or something similar?

[image]

I also noticed that the co2 ppm started to get really noisy as the temperature rise was happening:

[image]

keenanjohnson, 2021-09-16:
Perhaps the best compromise would be to just leave the enclosure white and keep the headpiece as green like this: 
[image]

keenanjohnson, 2021-09-16:
I set up another side-by-side comparison today just to get some additional data. I placed identical sensors in the normal white enclosure vs the green one as shown below. The sun is shining directly at these sensors right now.

[image]

There is a temperature rise above ambient in both cases, but the difference seems to be quite small. The blue curve below is the green sensor and the purple curve is the white sensor. The biggest difference seems to only be about 0.5 C. Given this, it's hard for me to believe that the green paint is really affecting the data in a negative way. Perhaps it's just best to advise that the sensor is placed in as shady an area as possible?

Does this conclusion make sense to you @spestana?

[image]

spestana, 2021-09-16:
> @spestana looking at one of the sensors from this morning, it looks like yours actually went offline around the peak temperature. Is that because you powered off the sensor or something similar?

Yes, it was unplugged for a while as I was seeing if I could move it elsewhere.

> I also noticed that the co2 ppm started to get really noisy as the temperature rise was happening:

I'm guessing rapid changes in T or RH would cause something like this, since the co2 measurements are calculated in part with T and RH.

> Perhaps it's just best to advise that the sensor is placed in as shady an area as possible?

Yes, shade (like the north side of a building in the N hemisphere) would be ideal.  Also don't want to be too close to the ground either (1-2 meters high).

spestana, 2021-09-16:
> Perhaps the best compromise would be to just leave the enclosure white and keep the headpiece as green like this:

Honestly, if they're kept in the shade, my guess is that this heating problem won't be an issue with the enclosures being green.

They do look nice all green :frog:

spestana, 2021-09-20:
I also wanted to note that the `temperature_offset` might not be necessary with a new enclosure that separates the SCD30 and the RaspberryPi (the very awesome design I see in #18 @eaudiffred )

keenanjohnson, 2021-09-21:
Indeed!

keenanjohnson, 2021-09-29:
Closing out for now given that we've updated the instructions and will move to a 3D printed enclosure which will eliminate the painting.

closed 2021-09-29 by keenanjohnson