Ribbit-Network/ribbit-network-frog-hardware #62: Replace Raspberry Pi With Alternate Linux Computer Due to Supply Chain Shortages

keenanjohnson, 2021-11-02:
Hello all! As you know there's a lot of global electronics shortages right now. This has caused me a lot of hassle in purchasing the Raspberry Pi's specifically, but I've always been able to find a few to build Frog Sensors. However, as of yesterday, it seems IMPOSSIBLE to buy any Raspberry Pi version in the US! This means we can't build any more Frogs in the short term. I've spoken with industry experts who says that the Raspberry Pi won't be reliably purchase-able until May 2022 at the earliest. 

SOOOOO this means we have a great opportunity / we must switch the design to an alternative compute module, accelerating work we wanted to do anyway for the solar power project. This work will probably be my focus from now until it's done haha. Let me know if you are interested in joining a working group over the next few weeks to quickly identify the right solution and get a design together.

**Research**
We created a spreadsheet outlining different options here: https://docs.google.com/spreadsheets/d/19OnLVaTxuk_TeqK9FWcrVBG1N89vUlChtp_3a1pAf28/edit?usp=sharing

The top two options seem to be:
* An ESP32 device which would be nice for it's low power capabilities, but makes firmware updating challenging since we can't use Balena
* A Beaglebone Black - This is a full Linux computer that allows us to use Balena and has a very large stock available.

**Solution**

I've engaged with the team at [Golioth.io](https://golioth.io/) to test their fleet update solution and it seems like their solution, while promising, is still not quite ready (See https://github.com/zephyrproject-rtos/zephyr/pull/37872) So we are going to proceed to design a Frog Sensor V2 based around the Beaglebone Black.

**Tasks**
- [ ] Revise the BOM
- [ ] Test the new electronics
- [ ] Update software to allow for deployment on the new platform
- [ ] Redesgin the enclosure for the new electronics

keenanjohnson, 2021-11-03:
I'm thinking instead of sticking a USB out of the top and making the enclosure really tall, it might make sense to use an extension cable and do something like this instead. Then we can stick the USB dongle next to the board, sort of like we did with the antenna previously: 
[image]

keenanjohnson, 2021-11-06:
I've put together a prototype sensor using the Beagle Bone Black and tried to run the Frog Software on it to see what immediate issues we will encounter. From the Circuit Python Board support file, these are the I2C pins

https://github.com/adafruit/Adafruit_Blinka/blob/f8c28157c2ffceaa436f106f0bbc4678b889cc39/src/adafruit_blinka/board/beagleboard/beaglebone_black.py#L123

keenanjohnson, 2021-11-08:
All right I haven't yet been able to get the SCD30 Python library working on the Beaglebone Black yet because I can't seem to get the dependencies correctly installed. [See this issue for more info.](https://github.com/adafruit/adafruit-beaglebone-io-python/issues/356)

djgood, 2021-11-19:
@keenanjohnson I suspect this issue _might_ have to do with the fact that you're running Python 3.10 in your container. The Adafruit_BBIO module works ok on my Beaglebone running Python 3.7.3. Although, Python and the Adafruit_BBIO module were pre-installed in my Beaglebone software image, so I'm not sure if they had to do something fancy to build it correctly.

Can you try to specify Python 3.7 in your `Dockerfile`? Change the first line to `FROM balenalib/%%BALENA_MACHINE_NAME%%-python-3.7`

I'm going to try upgrading my Python to 3.10 and see if I can reproduce your issue.

keenanjohnson, 2021-11-19:
I gave it a shot, but it looks like that image does not exist.

```
[main]     Step 1/8 : FROM balenalib/beaglebone-black-python-3.7
[main]     pull access denied for balenalib/beaglebone-black-python-3.7, repository does not exist or may require 'docker login': denied: requested access to the resource is denied
[Error]    Some services failed to build:
[Error]      Service: main
[Error]        Error: pull access denied for balenalib/beaglebone-black-python-3.7, repository does not exist or may require 'docker login': denied: requested access to the resource is denied
[Info]     Built on arm04
[Error]    Not deploying release.
Remote build failed
```

keenanjohnson, 2021-11-19:
Actually, it turns out I just needed a colon: FROM balenalib/beaglebone-black-python:3.7

Building now and will report back soon.

keenanjohnson, 2021-11-19:
Yeah unfortunately the same error. @djgood I've just given you access to a Balena fleet with my Beaglebone in it if you'd like to poke around.

```
[main]      ---> Running in be1e25d08e7e
[main]     Collecting Adafruit_BBIO
[main]       Downloading Adafruit_BBIO-1.2.0.tar.gz (78 kB)
[main]     Using legacy 'setup.py install' for Adafruit-BBIO, since package 'wheel' is not installed.
[main]     Installing collected packages: Adafruit-BBIO
[main]         Running setup.py install for Adafruit-BBIO: started
[main]         Running setup.py install for Adafruit-BBIO: finished with status 'error'
[main]         ERROR: Command errored out with exit status 1:
[main]          command: /usr/local/bin/python3.7 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-6k9luw4k/adafruit-bbio_b77f728ac826463ebd78d14f421c2e3d/setup.py'"'"'; __file__='"'"'/tmp/pip-install-6k9luw4k/adafruit-bbio_b77f728ac826463ebd78d14f421c2e3d/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-usmybr4m/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python3.7m/Adafruit-BBIO
[main]              cwd: /tmp/pip-install-6k9luw4k/adafruit-bbio_b77f728ac826463ebd78d14f421c2e3d/
[main]         Complete output (37 lines):
[main]         running install
[main]         running build
[main]         running build_py
[main]         creating build
[main]         creating build/lib.linux-armv7l-3.7
[main]         copying Adafruit_I2C.py -> build/lib.linux-armv7l-3.7
[main]         creating build/lib.linux-armv7l-3.7/overlays
[main]         copying overlays/builder.py -> build/lib.linux-armv7l-3.7/overlays
[main]         copying overlays/__init__.py -> build/lib.linux-armv7l-3.7/overlays
[main]         creating build/lib.linux-armv7l-3.7/Adafruit_BBIO
[main]         copying Adafruit_BBIO/sysfs.py -> build/lib.linux-armv7l-3.7/Adafruit_BBIO
[main]         copying Adafruit_BBIO/Encoder.py -> build/lib.linux-armv7l-3.7/Adafruit_BBIO
[main]         copying Adafruit_BBIO/__init__.py -> build/lib.linux-armv7l-3.7/Adafruit_BBIO
[main]         running build_ext
[main]         building 'Adafruit_BBIO.GPIO' extension
[main]         creating build/temp.linux-armv7l-3.7
[main]         creating build/temp.linux-armv7l-3.7/source
[main]         gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -DBBBVERSION41 -Isource/include/ -I/usr/local/include/python3.7m -c source/py_gpio.c -o build/temp.linux-armv7l-3.7/source/py_gpio.o -Wall
[main]         gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -DBBBVERSION41 -Isource/include/ -I/usr/local/include/python3.7m -c source/event_gpio.c -o build/temp.linux-armv7l-3.7/source/event_gpio.o -Wall
[main]         gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -DBBBVERSION41 -Isource/include/ -I/usr/local/include/python3.7m -c source/c_pinmux.c -o build/temp.linux-armv7l-3.7/source/c_pinmux.o -Wall
[main]         gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -DBBBVERSION41 -Isource/include/ -I/usr/local/include/python3.7m -c source/constants.c -o build/temp.linux-armv7l-3.7/source/constants.o -Wall
[main]         gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -DBBBVERSION41 -Isource/include/ -I/usr/local/include/python3.7m -c source/common.c -o build/temp.linux-armv7l-3.7/source/common.o -Wall
[main]         gcc -pthread -shared build/temp.linux-armv7l-3.7/source/py_gpio.o build/temp.linux-armv7l-3.7/source/event_gpio.o build/temp.linux-armv7l-3.7/source/c_pinmux.o build/temp.linux-armv7l-3.7/source/constants.o build/temp.linux-armv7l-3.7/source/common.o -L/usr/local/lib -lpython3.7m -o build/lib.linux-armv7l-3.7/Adafruit_BBIO/GPIO.cpython-37m-arm-linux-gnueabihf.so
[main]         /usr/bin/ld: build/temp.linux-armv7l-3.7/source/constants.o:/tmp/pip-install-6k9luw4k/adafruit-bbio_b77f728ac826463ebd78d14f421c2e3d/source/constants.h:4: multiple definition of `high'; build/temp.linux-armv7l-3.7/source/py_gpio.o:/tmp/pip-install-6k9luw4k/adafruit-bbio_b77f728ac826463ebd78d14f421c2e3d/source/constants.h:4: first defined here
[main]         /usr/bin/ld: build/temp.linux-armv7l-3.7/source/constants.o:/tmp/pip-install-6k9luw4k/adafruit-bbio_b77f728ac826463ebd78d14f421c2e3d/source/constants.h:5: multiple definition of `low'; build/temp.linux-armv7l-3.7/source/py_gpio.o:/tmp/pip-install-6k9luw4k/adafruit-bbio_b77f728ac826463ebd78d14f421c2e3d/source/constants.h:5: first defined here
[main]         /usr/bin/ld: build/temp.linux-armv7l-3.7/source/constants.o:/tmp/pip-install-6k9luw4k/adafruit-bbio_b77f728ac826463ebd78d14f421c2e3d/source/constants.h:7: multiple definition of `output'; build/temp.linux-armv7l-3.7/source/py_gpio.o:/tmp/pip-install-6k9luw4k/adafruit-bbio_b77f728ac826463ebd78d14f421c2e3d/source/constants.h:7: first defined here
[main]         /usr/bin/ld: build/temp.linux-armv7l-3.7/source/constants.o:/tmp/pip-install-6k9luw4k/adafruit-bbio_b77f728ac826463ebd78d14f421c2e3d/source/constants.h:6: multiple definition of `input'; build/temp.linux-armv7l-3.7/source/py_gpio.o:/tmp/pip-install-6k9luw4k/adafruit-bbio_b77f728ac82646 [... cut ...] 78d14f421c2e3d/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-usmybr4m/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python3.7m/Adafruit-BBIO Check the logs for full command output.
```

djgood, 2021-11-19:
> Actually, it turns out I just needed a colon: FROM balenalib/beaglebone-black-python:3.7
> 
> Building now and will report back soon.

Oops my bad, sorry.

And shoot, I was hoping it was a quick fix. Looks like the downgrade did clear up some of the warnings, at least. 

I'll see what I can figure out from building on the device.

djgood, 2021-11-19:
Good news: Looks like the maintainers of Adafruit_BBIO have a fix for this issue. Bad news: They haven't cut a release containing the fix yet. Relevant commit [here](https://github.com/adafruit/adafruit-beaglebone-io-python/commit/be710eaf9bbcd232cc627e7a3e7073fdc47fb9ac).

So I think the underlying issue is due to compiling with a newer version of gcc included in the most recent version of Debian (11.x) used by balenalib/beaglebone-black-python. The official Beaglebone images are running Debian 10.x, so they must be using an older version of gcc that doesn't throw this error.

Two solutions I can think of:
1. Use a `balenalib/beaglebone-black-python:buster` image.
2. Download the master branch for Adafruit_BBIO and compile/install the newest version ourselves to get the fix.

I'm going to try both approaches and see what I can get working. Currently learning toward keeping our image OS distro versions in line with the official Beaglebone ones so we don't hit this kind of error in the future.

keenanjohnson, 2021-11-19:
Sounds great! Thanks and keep us posted!

djgood, 2021-11-19:
Quick test to verify everything is running smoothly (without GPS attached):

```
Python 3.10.0 (default, Oct  7 2021, 02:56:46) [GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import adafruit_scd30
>>> import board
>>> i2c_bus = board.I2C()
>>> scd = adafruit_scd30.SCD30(i2c_bus)
>>> scd.ambient_pressure = 1007
>>> scd.temperature_offset = 4
>>> scd.altitude = 0
>>> scd.self_calibration_enabled = True
>>> scd.data_available
1
>>> (scd.CO2, scd.temperature, scd.ambient_pressure)
(464.842041015625, 20.303058624267578, 1007)
>>> import adafruit_dps310
>>> dps310 = adafruit_dps310.DPS310(i2c_bus)
>>> (dps310.pressure, dps310.temperature)
(1020.401282227745, 22.273026113435037)
```

And we plugged in the GPS and everything worked!

I think the best approach is to go with the older Debian image as long as this doesn't disrupt things on the Raspberry Pi fleet. This is for the reason I mentioned above, most code designed to run on the Beaglebone is designed for this specific version of Debian. So tracking the same version will hopefully avoid any dependency build issues in the future. I'll push a test build to my RPi and see if this version of Debian breaks anything.

keenanjohnson, 2021-11-20:
The beaglebone black i2c driver supposedly supports clocking stretching. Going to run the test Beaglebone black for a day or so to verify we don't initially see this error: https://github.com/Ribbit-Network/ribbit-network-frog-sensor/issues/61

keenanjohnson, 2021-11-22:
Ran the beaglebone for over 48 hours and didn't see any i2c issues, so I think we can call that good. I'm going to revise the BOM and documentation today for our release of this sensor version.

closed 2021-12-07 by keenanjohnson