Ribbit-Network/ribbit-network-frog-hardware #101: Influx KeyError in co2 service

alanb128, 2022-01-22:
Using a Raspberry Pi 4 2GB, balenaOS 2.88.5+rev1. On first boot after pushing the application with the CLI:
```
 co2  Traceback (most recent call last):
 co2    File "/usr/src/co2.py", line 320, in <module>
 co2      main()
 co2    File "/usr/src/co2.py", line 229, in main
 co2      client = InfluxDBClient.from_config_file("influx_config.ini")
 co2    File "/usr/venv/lib/python3.10/site-packages/influxdb_client/client/influxdb_client.py", line 170, in from_config_file
 co2      url = config_value('url')
 co2    File "/usr/venv/lib/python3.10/site-packages/influxdb_client/client/influxdb_client.py", line 168, in config_value
 co2      return config['influx2'][key].strip('"')
 co2    File "/usr/local/lib/python3.10/configparser.py", line 964, in __getitem__
 co2      raise KeyError(key)
 co2  KeyError: 'influx2'
```

keenanjohnson, 2022-01-22:
Hello @alanb128 ! Sorry for that erorr! I think you are probably the first person besides myself to push to a device instead of just downloading the balena image :).

That error is because the influxdb config file is missing. 

The script is looking for a file called influx_config.ini in the software/co2 directory. You've made me realise this is missing from the readme, so I'll add that in.

That file has the format below. I don't commit the file to keep the token out of the public domain for the public database. 

I'd love to support you in whatever you are trying to do! If you are trying to build a unit to test out some improvements for the code for example, you can easily start your own free trial of influx and add your own key in the format below!

Or if you are trying to build a device to contribute, I would recommend joining our [balena open-fleet](https://hub.balena.io/g_keenan_johnson1/ribbit-network)! If you join the fleet, you'll download an image preconfigured with everything and your device will follow the software updates with the rest of the fleet as we build out the network.

Hope that's helpful and don't be a stranger in the Discussions here on Github or in our Discord server!

```
[influx2]
url=https://us-west-2-1.aws.cloud2.influxdata.com
org=keenan.johnson@gmail.com
token=
timeout=6000
verify_ssl=True
```

alanb128, 2022-01-22:
Thanks for the quick response @keenanjohnson ! No problem, I used the CLI because I could not get the Hub image to work. I tried two different Raspberry Pi 4s and different WiFi networks but I just get the flashing LED four times. Once or twice I was able to get the WiFi Connect menu, but even entering credentials into that did not connect me. It's quite possible I'm missing something obvious but strange that it happens on two different devices. Ethernet works fine, but I can't place it outside using Ethernet!

keenanjohnson, 2022-01-23:
I see, sorry you were having so much trouble getting it to work! A few questions to help me figure out how best to help you:

* You are using a Raspberry Pi 4, not a CM4 correct? 
* Are you using a USB wifi adapter or the on-board wifi?
* Did you enter your wifi credentials when you tried to download the image from the hub?

alanb128, 2022-01-23:
- Yes, using a Rapsberry Pi 4 2GB. (I plan to switch to a CM4 soon, but I wanted to get started with what I had available.)
- I tried both the on-board WiFi and the USB WiFi adapter.
- Yes, I entered credentials when downloading the Hub image.

alanb128, 2022-01-23:
Just did some more testing and it seems like unplugging the GPS USB module allows the Pi to connect...

keenanjohnson, 2022-01-23:
I see! Thanks for the info!

Hard to say exactly what happened, but my intuition is that the USB cable is interfering with the Wifi reception strength. Many USB cables are poorly shielded and radiate quite a lot of RF energy which can disrupt weak wifi antennas. And the onboard antennae on the pi four is very weak unfortunately, that's why our design uses the CM4 with external antennae or the Beaglebone with external antennae.

You should have had better results with the USB dongle, but depending on which you used, the reception might not be much better, unfortunately. Additionally, Balena doesn't support some of the common wifi adapters, unfortunately, which could also be the source of issues.

alanb128, 2022-01-23:
Thanks, I hope to switch to a CM4 soon. In the meantime, will the project work without the GPS module? I don't see my device on the Ribbit Network map.

keenanjohnson, 2022-01-23:
Unfortunately not at the moment, the sensor can't locate itself on the map without the GPS module telling it where it is.

Perhaps in the future the software could be improved to allow someone to manually input a sensor location, but no one has built that feature yet. I'm not sure of your skill set, but it could be something you take on :)

alanb128, 2022-01-23:
I wonder if you could use the balena API to get location? https://www.balena.io/docs/reference/api/resources/device/

keenanjohnson, 2022-01-23:
That would be possible @alanb128! Just from my experience looking at the map in Balena, I have seen the balena reported location based on IP address to be off by 10s or hundreds of kilometers. Our target for accuracy is to have ~ a 1km accuracy, so I think having a user manually place the sensor in the event of a lack of GPS would be more accurate.

alanb128, 2022-01-24:
It looks like once the device boots over WiFi, I can plug in the USB GPS module and it stays online. (At least I don't get the four LED flashes.) Can you confirm if my device is online? I switched to the Hub image so I don't think I have that visibility.

keenanjohnson, 2022-01-24:
Are you located in Pnnsylvania @alanb128 ? If so, I see a new sensor online there. That sensor is lacking a GPS fix at the moment, which is why it is not located on the map yet.

keenanjohnson, 2022-01-24:
The GPS sensor will need to be outside and have a clear view of the sky to work. You can see some information about this in the debugging guide we have here: https://github.com/Ribbit-Network/ribbit-network-frog-sensor/blob/main/hardware/v2/assembly-instructions/7-debugging.md

"You can verify this by checking the LED on the GPS module. If the LED is blinking then a GPS location fix is established and everything is great. However, if the light on the GPS module is solid, then the module cannot establish a fix. Try moving the sensor to a less obstructed area."

alanb128, 2022-01-24:
Yep, that's me in Pennsylvania. I'm surprised that the GPS can't get a fix. It is located outside with no case and the antenna pointed at the sky. Maybe a bad module? Anyway, my CM4 carrier board arrived so I'll switch to that and see how it goes...

keenanjohnson, 2022-01-24:
Which gps module are you using @alanb128 ? The one from the V1 or V2 version of the bom? The version one sensor can have quite variable performance, though it is significantly cheaper. One thing to watch out for if you are using the V1 sensor is to make sure the GPS ant. cable is not touching anything else conductive or radiating. For example, if it is touching the USB cable, then that can often destroy the GPS reception it seems. The V2 module is much more tolerant to noise I found.

alanb128, 2022-01-25:
I'm using the V1 GPS. I'll try re-routing the antenna, but it's a bit crowded in the V1 bracket.
Also from the V1 BOM, I'm using the base board, No WiFi CM4 (although with 16GB eMMC) and TP-Link USB adapter. I can't seem to get WiFi working at all on this setup. Based on [this page](https://www.waveshare.com/wiki/CM4-IO-BASE-A) I tried adding `dtoverlay=dwc2,dr_mode=host` to the config.txt but that didn't seem to help. I would try to troubleshoot but it's tough without access!

Anyway, it's sitting on my roof temporarily using Ethernet to connect and the GPS module antenna spread out with a clear view of the sky...

keenanjohnson, 2022-01-25:
Yeah definitely agree that the V1 bracket isn't optimal, we tried to fix all of those issues in V2. 

Alan, Balena edits the config.txt based on the balena dt_overlay env variable, so editing it by hand probably won't make a difference as the Balena supervisor will probably overwrite your hand edits. However, if you are on the balena Ribbit-Network fleet, the dwc2,dr_mode=host items should always be there.

Can you share which TP-Link Wifi adapter you have? Also, if you put the device online, I can ssh in and poke around to see if I can sus out what the issue is more specifically.

alanb128, 2022-01-25:
Thanks for all of your help @keenanjohnson , it would be good to get this working.

Just FYI, I tried adding the config.txt [before the first boot](https://www.balena.io/docs/reference/OS/advanced/#modifying-configtxt-locally-before-the-first-boot) so it would not be overwritten.

I have the TL-WN725N, FCCID: TE7WN725N. The device is online now via Ethernet, with the GPS module separated from the bracket and the antenna aimed at clear, open sky. The wireless adapter is also plugged in.

keenanjohnson, 2022-01-25:
Ok Alan!

I took a look and noticed a few things:

## GPS 
I first looked at your GPS reception. Interestingly, it seems as if your GPS unit has visibility to enough satellites, but it is reporting a location of 0 Lat, 0 Long. See the output directly from the GPS below (cmd: cgps -s ).

I've never seen that behavior before and would explain why the sensor isn't showing up on the map yet. We filter any 0,0 positions as that is likely to be a device in some sort of test mode or in an error condition. 

I checked the serial output of the GPS device manually and it is indeed reporting a null location, which points to some sort of firmware issue internal to the GPS I would imagine. The best advice I have for you there is to restart power to the GPS and hopefully that will resolve that? If not, this may be a faulty GPS module in some way. 

```
┌───────────────────────────────────────────┐┌──────────────────Seen 16/Used  9┐
│ Time:        2022-01-25T20:36:53.000Z (18)││GNSS   PRN  Elev   Azim   SNR Use│
│ Latitude:          0.00000000 N           ││GP  4    4  19.0  193.0  25.0  Y │
│ Longitude:         0.00000000 E           ││GP  7    7  56.0  310.0  34.0  Y │
│ Alt (HAE, MSL):      0.000,    -17.000 m  ││GP  8    8  82.0   93.0  25.0  Y │
│ Speed:             0.00 km/h              ││GP  9    9  35.0  229.0  26.0  Y │
│ Track (true, var):     0.0,  -4.8     deg ││GP 14   14   5.0  269.0  27.0  Y │
│ Climb:             0.00 m/min             ││GP 16   16  25.0   70.0  27.0  Y │
│ Status:         3D FIX (5 secs)           ││GP 21   21  25.0  146.0  22.0  Y │
│ Long Err  (XDOP, EPX):  0.58, +/-  8.8 m  ││GP 27   27  48.0   49.0  31.0  Y │
│ Lat Err   (YDOP, EPY):  0.68, +/- 10.2 m  ││GP 30   30  28.0  310.0  42.0  Y │
│ Alt Err   (VDOP, EPV):  1.42, +/-544967 m ││GP  1    1   5.0  168.0   0.0  N │
│ 2D Err    (HDOP, CEP):  0.90, +/-  0.0 m  ││SB133   46  38.0  213.0   0.0  N │
│ 3D Err    (PDOP, SEP):  1.68, +/- 1899 m  ││SB135   48  16.0  248.0   0.0  N │
│ Time Err  (TDOP):       0.87              ││SB138   51  33.0  224.0   0.0  N │
│ Geo Err   (GDOP):       1.89              ││QZ  2  194   n/a    0.0   0.0  N │
│ ECEF X, VX:    1242780.220 m    0.010 m/s ││QZ  3  195   n/a    0.0   0.0  N │
│ ECEF Y, VY:   -4731859.050 m    0.000 m/s ││QZ  4  196   n/a    0.0   0.0  N │
│ ECEF Z, VZ:    4078549.520 m   -0.030 m/s ││                                 │
│ Speed Err (EPS):       +/-  1.4 km/h      ││                                 │
│ Track Err (EPD):        n/a               ││                                 │
│ Time offset:            0.109661066 s     ││                                 │
│ Grid Square:            JJ00aa00          ││                                 │
└───────────────────────────────────────────┘└─────────────────────────────────┘
```

## Wifi Interface
From what I can see the wifi interface is working just fine! You can see the network manager output below which shows an active connection on both ethernet and wifi:

```
root@b3cb783:/mnt/boot/system-connections# nmcli device
DEVICE           TYPE      STATE                   CONNECTION         
eth0             ethernet  connected               Wired connection 1 
wlan0            wifi      connected               balena-wifi-01     
supervisor0      bridge    connected (externally)  supervisor0        
balena0          bridge    unmanaged               --                 
br-c0678580792d  bridge    unmanaged               --                 
resin-dns        bridge    unmanaged               --                 
vethc6fac4f      ethernet  unmanaged               --                 
lo               loopback  unmanaged               --                 
resin-vpn        tun       unmanaged               --                 
root@b3cb783:/mnt/boot/system-connections# nmcli dev wifi list
IN-USE  BSSID              SSID                             MODE   CHAN  RATE       SIGNAL  BARS  SECURITY 
*       B6:FB:E4:DA:09:E2  M637T                            Infra  6     16 Mbit/s  69      ***   WPA2     
        E6:FB:E4:DA:09:E2  --                               Infra  6     16 Mbit/s  68      ***   WPA2     
        C6:FB:E4:DA:09:E2  M637G                            Infra  6     16 Mbit/s  68      ***   WPA2     
        C8:A7:0A:A3:32:C8  FiOS-IZ824                       Infra  1     16 Mbit/s  58      ***   WPA2     
        70:5A:0F:F1:F8:8E  DIRECT-11-HP OfficeJet Pro 8710  Infra  1     44 Mbit/s  48      **    WPA2     
root@b3cb783:/mnt/boot/system-connections# 
```

alanb128, 2022-01-25:
I just power cycled the GPS so we can see if that helps! Once that's resolved, I'll disconnect the Ethernet. I did notice the dongle LED is flashing now which it hadn't before. I didn't change anything about that though!

keenanjohnson, 2022-01-25:
Hmmm looks like your device is offline now.

alanb128, 2022-01-25:
That's strange, it hasn't been touched and is still on Ethernet. I power cycled the whole device just now.

keenanjohnson, 2022-01-25:
Weird.

Well, it seems as if everything is working now!

[image: image.png]

alanb128, 2022-01-25:
Great! Ok, final test: I just disconnected the Ethernet cable, so it's on WiFi only now.

alanb128, 2022-01-26:
Looks like GPS and WiFi are working now @keenanjohnson . The device is outside and the data is appearing on your map. I'm not sure what changed - I didn't really modify anything from my initial build but I'm glad it's working. Thanks for talking me through it. You can close this issue, but I'll be in touch in the future - I have a few suggestions!

keenanjohnson, 2022-01-26:
Excellent! Glad it's all working @alanb128 ! I've taken a few notes about improving the docs and I'll fix some of those on this issue and then close it out!

Yes! Please be in touch, we would love to hear your experiences on the build and ideas for how to make it better! Feel free to join our discord or start a Github discussion! 

The most obvious thing with your build I believe is to prioritize the creation of a local UI so that you can self-diagnose issues with your sensor. We've been tracking this for a while, but haven't had anyone step up to develop it. I'm actively searching for volunteers that might want to take it on: https://github.com/Ribbit-Network/ribbit-network-frog-sensor/issues/57

closed 2022-01-26 by keenanjohnson