Ribbit-Network/ribbit-network-frog-hardware #103: Research Options for Methane Sensor

keenanjohnson, 2022-01-27:


josuah, 2023-04-18:
Going for a web search brings up a lot of Aliexpress links featuring a sensor witten MQ-4 on it.

Further search about that MQ-4 leads us to this:
https://www.sparkfun.com/products/9404
https://www.sparkfun.com/datasheets/Sensors/Biometric/MQ-4.pdf

> 200 to 10000ppm
Is 200 part per million sensible enough?

The manufacturer seems oriented toward practical uses for these, such as detecting "anything hazardous" rather than one particular gas:
https://www.hwsensor.com/

Yet, an inventory search for that parts reveals the category with many other similar parts, and here we have our full range of gas sensors, this is the right place with all kind of specs

https://www.digikey.fr/en/products/filter/gas-sensors/530
https://eu.mouser.com/c/sensors/environmental-sensors/air-quality-sensors/
https://www.arrow.com/en/categories/sensors/air-quality-sensors

Maybe that can lead to one sensor having multiple kind of gas, barometer... all included.
But at least we can compare now.

keenanjohnson, 2024-01-05:
@hansenni let's use this issue to track your work to create a sensor comparison matrix as we discussed today.

hansenni, 2024-01-05:
@keenanjohnson sounds like a plan. I'll start with CO2 sensors and head towards methane. After that we can discuss particulate, VOC, or anything else of interest

keenanjohnson, 2024-01-23:
Hey @hansenni. How is this going?

hansenni, 2024-01-23:
On track to finish CO2 sensors this week I believe. Work got a bit busier this week or it would already be done.

hansenni, 2024-01-24:
| Sensor                 | Technology                                   | Measurement Range (ppm)                        | Accuracy                                                                                                                                                                                                       | Supply Voltage                                                                              | Average Current                              | Interface               | Dimensions                     | Operating Temperature Range | Price (if available)         | Link                                                                                                                                                                                                                                       |
| ---------------------- | -------------------------------------------- | ---------------------------------------------- | -------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- | ------------------------------------------------------------------------------------------- | -------------------------------------------- | ----------------------- | ------------------------------ | --------------------------- | ---------------------------- | ------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------ |
| Senseair K33 ELG       | NDIR                                         | 400-5000 ppm                                   | ± (30 ppm + 3%MV)                                                                                                                                                                                              | 4.75-12VDC 3 maximum rating, powered via Vbat+<br>5.5-12VDC3 maximum rating, powered via G+ | <52uA (1 measurement/hour)                   | UART, I2C               | 11.80mm x57.15x50.80           | 0-50 °C                     | ~180 USD                     | [https://www.co2meter.com/products/k33-environmental-logger-co2-sensor](https://www.co2meter.com/products/k33-environmental-logger-co2-sensor)                                                                                             |
| Senseair K30           | NDIR                                         | 0-5000 ppm Up to 10000 ppm extended range      | ± (30 ppm +3% of reading)1 (Extended range+10% of reading)                                                                                                                                                     | 4.5 - 14.0 V DC maximum rating                                                              | 40 mA average                                | Analog, UART            | 51 x 58 x 12                   | 0-50 °C                     | ~78 USD                      | [https://www.co2meter.com/products/k-30-co2-sensor-module](https://www.co2meter.com/products/k-30-co2-sensor-module)                                                                                                                       |
| MH-Z19                 | NDIR                                         | 0 – 0.5% VOL optional                          | ± (50ppm + 5% reading value)                                                                                                                                                                                   | 3.6 ~ 5.5 V DC                                                                              | < 18 mA                                      | UART, PWM               | 33 mm x 20 mm x 9 mm (LxWxH)   | 0 ~ 50 ℃                    | ~34 USD                      | [https://www.amazon.com/MH-Z19-MH-Z19B-Sensor-Infrared-0-5000ppm/dp/B07VCS9VBM](https://www.amazon.com/MH-Z19-MH-Z19B-Sensor-Infrared-0-5000ppm/dp/B07VCS9VBM)                                                                             |
| SCD40/41               | Photoacoustic NDIR sensor technology PASens® | SCD40: 400 – 2,000 ppm, SCD41: 400 – 5,000 ppm | SCD40: ±(50 ppm + 5% of reading) for 400 – 2,000 ppm; SCD41: ±(50 ppm + 2.5% of reading) for 400 – 1,000 ppm, ±(50 ppm + 3% of reading) for 1,001 – 2,000 ppm, ±(40 ppm + 5% of reading) for 2,001 – 5,000 ppm | 2.4 – 5.5 V                                                                                 | 5 sec: 11 – 15 mA, 30 sec: 3.2 – 3.5 mA (5V) | Digital I2C             | 10.1 x 10.1 x 6.5 mm3          | \-10 – 60 °C                | SCD41: ~50USD, SCD40: ~45USD | [https://www.adafruit.com/product/5190](https://www.adafruit.com/product/5190)                                                                                                                                                             |
| SCD30                  | NDIR CO2 sensor technology                   | 400 ppm – 10,000 ppm                           | ± (30 ppm + 3%MV)                                                                                                                                                                                              | 3.3 V – 5.5 V                                                                               | 19 mA @ 1 meas. per 2 s.                     | UART, I2C               | 35 mm x 23 mm x 7 mm  [... cut ...]  | 4.5-5.25V                                                                                   | 300mA peak, 30mA average                     | UART, Modbus            | 33.9 x 19.8 x 8.7 mm3          | 0 – 50 ℃                    | ~47 USD                      | [https://www.co2meter.com/products/004-0-0010-senseair-s8-commercial-co2-sensor](https://www.co2meter.com/products/004-0-0010-senseair-s8-commercial-co2-sensor)                                                                           |