Ribbit-Network/ribbit-network-frog-hardware #126: Create a Golioth ESP32 Frog Sensor Prototype

keenanjohnson, 2022-05-27:
As discussed in #63, one of the main challenges in creating a microcontroller based Frog Sensor is establishing a software update system that works as well as [Balena](https://www.balena.io/) does with our current single board linux compute options.

One promising new option is https://golioth.io/. I have been in contact with their team and they are eager to support us! The first step is to create a prototype of the sensor with their software stack and an ESP32.

**Steps**
* Get an ESP32 and the relevant sensor modules from the BOM (CO2, Barometer, GPS)
* Build a prototype software application using Golioth that supports our sensors
* Test pushing software updates to that device via Golioth

Golioth has a free trial which should support everything that is needed.

Aravind-Krishnadeva, 2022-05-29:
Greetings, Keenan, Was considering, OTA updates, and nature of our sensors too, when going for a solar powered option. Was just trying to figure out, how frequent will be the updates ( or infrequent), when planning for a solar module. ..

heatblast016, 2022-05-30:
Updating this -- I have the Golioth 'Hello' example working on an ESP32 dev board. Next step is assembling it with NDIR/Barometer and getting a software application working

keenanjohnson, 2022-06-24:
Related to this, I created this issue to build out the open-fleet onboarding process for Golioth devices: https://github.com/Ribbit-Network/ribbit-network-frog-sensor/issues/128

keenanjohnson, 2022-07-27:
Golioth (https://golioth.io/) just made it easier to add a couple of cloud functions to an ESP32 without having to become an AWS or Azure expert. We're also going to continue building out training for ESP-IDF  https://blog.golioth.io/announcing-the-golioth-esp-idf-sdk/. Hope it is useful for future frogs!

beriberikix, 2022-07-29:
Basically, same platform as before (software updates, streaming sensor readings, APIs) but with even better support for ESP32 :)

beriberikix, 2022-08-09:
Hello! Saw the latest [Roadmap doc](https://ribbitnetwork.notion.site/8-4-Core-Team-Meeting-5f379fd414394d63aa79ad32dcf1c924) and was wondering where this falls on the priority. Thanks!

keenanjohnson, 2022-08-09:
My @beriberikix ! Good question! This issue has actually been derisked quite a bit and our main barrier before considering switching to an ESP32 is actually in ensuring that the Golioth fleet management will work with our open-source, open-fleet concept. See #128.

We don't have that tied to the roadmap as there is still a fairly large amount of technical and pricing risk for Ribbit to use Golioth.

So far I haven't found a developer volunteer that is excited and has time to work on that. If anyone at Golioth is interested in helping out, that would certainly accelerate this process and we would be able to think about how to integrate it into the roadmap.

keenanjohnson, 2022-09-26:
@heatblast016 did you ever get anywhere with this work? Is there source code anywhere we could pick up and start from?

keenanjohnson, 2022-09-30:
I've created both a milestone and a git branch to start breaking this work into small chunks and to start committing source code.

keenanjohnson, 2022-10-01:
I've added what I think are the basic files for a golioth app into this folder in the branch: https://github.com/Ribbit-Network/ribbit-network-frog-sensor/tree/esp32_golioth/software_esp32

keenanjohnson, 2022-10-18:
This issue has been turned into a milestone with many more tasks, so closing this down in favor of the milestone: https://github.com/Ribbit-Network/ribbit-network-frog-sensor/milestone/4

closed 2022-10-18 by keenanjohnson