Ribbit-Network/ribbit-network-frog-hardware #129: Investigate How Low Temperatures Affect the Sensors

eaudiffred, 2022-07-14:
There is a potential concern with low outside temperatures causing problems with sensor accuracy or even the ability to operate.
Below is a table of recommended operating temperatures for each component.  These numbers were taken from the data sheet for each component.  
[image]

eaudiffred, 2022-07-19:
https://www.adafruit.com/product/5187#technical-details
This sensor has a wider operating temp -10C to 60C which is better, but still not great.  I'd like to see the low end somewhere around -20C or -30C.  
It uses slightly less power and is also $10 cheaper than the SCD-30
The accuracy is slightly worse though.  5% compared to the SCD-30 at 3%
The range is also significantly less.  400ppm-2,000ppm vs 400ppm-10,000ppm on the SCD-30.  I don't think the range should be an issue though.

keenanjohnson, 2022-07-21:
Hey @eaudiffred ! Yeah we had seen that sensor, but decided to go with the SCD-30 as it has a slightly more accurate range. However, given that we have now seen temperature range issues with the SCD-30, it is probably worth digging into further.

eaudiffred, 2022-11-04:
It's starting to get cold here in Utah and I notice that my sensor is going in and out.  It doesn't perfectly correlate to the temperature swings each day, but I'm wondering if the cold is causing the problem.  

I'll try rebooting today.  It's only been below freezing a few times so far, so I'm surprised to see the sensor dropping out as much as it is.  

[image]
[image]

eaudiffred, 2022-12-14:
As soon as I can get a V4 sensor up and running I'll be interested to see if our low temp issues are resolved.  I believe the ESP32 has a lower temp operating range than the Raspberry Pi.

keenanjohnson, 2022-12-14:
yes agreed! We are still finishing out the setup software for the V4 sensor, but this will be quite interesting to test! /milestone/1

eaudiffred, 2023-02-23:
We are now running V4 of the Frog sensor.  I have one outside my home and will monitor this winter.