Ribbit-Network/ribbit-network-frog-hardware #130: Beaglebone Green Gateway is Out of Stock

keenanjohnson, 2022-07-22:
Sigh.... the Beaglebone Green Gateway is now out of stock. We originally pivoted to this as the Raspberry Pi CM4 was also out of stock everywhere. 

Time to figure out a solution.

keenanjohnson, 2022-07-22:
See this issue for the last switch: https://github.com/Ribbit-Network/ribbit-network-frog-sensor/issues/62

pascaljoly, 2022-07-24:
so we are back to the Raspberry Pi version?

keenanjohnson, 2022-07-25:
It seems likely yes. However, it also seems like the Raspberry Pi CM4 with Wifi is mostly out of stock everywhere. I have some stock of them here at Ribbit Network HQ, but it's unfortunate that they aren't buyable on the wider market. 

However, I have confidence and assurance from the Raspberry Pi foundation that the raspberry pi CM4 will be in production and support for a long time. The manufacturer of the Beaglebone Gateway has told me they aren't sure when they will produce more, so the way forward is a bit obvious in the Raspberry Pi Cm I think.

keenanjohnson, 2022-07-26:
I've created this milestone which will track all the work needed to get this version of the Frog up and running again.

https://github.com/Ribbit-Network/ribbit-network-frog-sensor/milestone/3

abesto, 2022-07-26:
This makes me wonder... do you expect that at some point there'll be some solution that can reliably support all the scale Ribbit needs, or do we lean into heterogeneous hardware?

keenanjohnson, 2022-07-26:
It's a great question to ask! 

Prior to the pandemic, the Raspberry Pi CM4 was in vast mass production and would have been totally sufficient for our needs, so I have sort of been assuming (after some conversations with the raspberry pi team) that this will return eventually.

However, that is a speculative assumption and it may be wiser to assume that this is not the case. Trying to figure out how to maintain both the mechanical part geometries and build instructions for several versions is the main limitation there, so I'm open to suggestions on the best way to accommodate that!

pascaljoly, 2022-07-26:
I believe since it is an open source project we will have to live with a few branches of the sensor. as we are already seeing, it will depend on supplies availability, and in the future some additional components (solar panel, CH4/NO2 sensors...).
As Keenan mentioned we should aim to have a universal enclosure, if possible, but even that might be far fetched.
 We do want to make sure the data that is uploaded is normalized so that we don't run into inconsistencies when we try to aggregate.

keenanjohnson, 2022-07-27:
Yeah in my mind it's just a question of priorities and when. I think the pain of maintaining multiple instruction sets currently outweighs the benefits, but it probably will not in the future!

If someone is passionate about how to organize build instructions and design files to support this, I would welcome them to take on the challenge!

closed 2022-08-23 by keenanjohnson