Ribbit-Network/ribbit-network-frog-hardware #131: Identify Ideal Supplier for 3D Printer Parts

keenanjohnson, 2022-07-22:
In Discord, we discussed that it would be ideal to identify one or more ideal suppliers for the 3d printed parts so that those without access to a 3D printer, could easily order the parts.

keenanjohnson, 2022-07-22:
Options posted in Discord:

* https://www.protolabs.com/
* https://www.xometry.com/
* https://www.hubs.com/
* https://www.shapeways.com/
* https://i.materialise.com/en
* https://www.sculpteo.com/en/
* https://www.fictiv.com/
* https://jlcpcb.com/3d-printing

Thanks @beriberikix

keenanjohnson, 2022-07-22:
@eaudiffred

eaudiffred, 2022-07-29:
I've looked through about 12 manufactures, some 3D printing and some injection molding.  

- [CraftCloud](https://craftcloud3d.com/) seems like a good one-off option.  FDM printing, material (PETG) and color selection (Green) are available.  The total cost of the four main pieces of the enclosure came to $78.83.  This is pretty much identical to what we are doing now.
- [ProtoLabs](protolabs.com) is in the process of reviewing our part files and will get back with me next week.  They can do 3D printing or injection molding.  
- I sent a request to [VistaTek](http://vistatek.com/) to review our part files and see if they are interested.  They are a large scale injection molding company.

eaudiffred, 2022-08-02:
Also in contact with [Fictiv](https://www.fictiv.com/) who is quoting us 3D printed parts as well as urethane injection molded parts.  The urethane mold is an interesting option.  Significantly cheaper than a steel mold, but it can only be used about 50 times.  At this point 50 units seems like plenty so I'll continue to explore that option.  Still working on the material choice. The color green and the priority of either a recycled or recyclable material are proving to be the limiting factors.  I think of both of those specification are worthwhile though.

eaudiffred, 2022-08-02:
Waiting on communication back from [Xometry](https://www.xometry.com/) and[ Shapeways ](https://www.shapeways.com/)as well.

eaudiffred, 2022-08-02:
Quick Update: 
Fictiv, ProtoLabs, and Shapeways don't have a competitive cost for the low volume of parts that we need right now.  I've been using a qty of 50 as a ballpark.  Recyclability and the color green were also more difficult to find than I expected.   The urethane casting molds were about $2k/ea.  The steel molds were around $5-6k/ea.  To me that kind of investment does not make sense right now.

eaudiffred, 2022-08-02:
CraftCloud is still looking like the best option for one-off ordering.  I have the four pieces of the enclosure, two eyes, and one mount quoted at about $85.  CraftCloud looks to be a liaison that is connecting consumers with various small scale 3D printing manufactures.

eaudiffred, 2022-08-05:
Looked into JLCPCB.  Unfortunately they only offer black ABS for 3D printing services right now.  They seem to focus on pcb manufacturing, so if that is something we need in the future they might be a good option.

eaudiffred, 2022-08-09:
I spoke with Xometry on the phone yesterday.  They recommended using their FDM services with green ASA material.  The parts range from $15-$50/ea.

keenanjohnson, 2022-08-09:
Interesting! Did they specific why ASA vs PETG?

eaudiffred, 2022-08-10:
They just don't offer PETG.  ASA is the only material they offer that has both outdoor resistance and color options.

eaudiffred, 2022-08-10:
The tricky thing is finding a supplier that is even remotely close in cost to what we can produce at home.  Currently the four large parts of the enclosure add to just under 250g of PETG and take approximately 24 hours to print.  That's about $6.25 in material and ~$0.40 in electricity.  

I certainly did not expect to find anything that would match our cost, but I was hoping for something in the $30-$40 range per enclosure.  So far the lowest cost option came in at $85.  

Still out there looking though!

keenanjohnson, 2022-08-10:
Thanks for the search! That's all useful information!

eaudiffred, 2022-08-11:
Hubs.com has pretty great discounts on bulk pricing, something I have not seen from other suppliers.  At an order qty of 25 the price per unit is pretty good!  In the online quoting software they don't offer color options though.  I'll have to look into that further.  

[image]

eaudiffred, 2022-09-08:
I've hit a wall with this one.  With one-off/low qty manufacturing the prices are high.  I think we have a decent amount of options when we reach the point of needing bulk qtys (1000+).  
I don't know what our limit is with the 3D printers we have right now, but in the short term, it seems like we could maintain a small inventory and sell the parts on the store for cost (~$7).  
It takes about 24hrs to print the four large parts.  I may be able to trim that down an little by optimizing the print settings.

keenanjohnson, 2022-09-08:
Yeah, thanks for digging into this @eaudiffred ! That answer seems pretty definitive to me.

To summarize the findings from my understanding:

* There is no cost-effective option for a 3D printing service as low volume (<1000 pieces)
* The most responsible short term solution is to maintain a Ribbit Network 3D printing farm and inventory and sell those as part of the kits until we reach high volumes (~1000 pieces)

It's a great learning and I'm glad we did the research to be sure we know the correct path. Does that match your understanding @eaudiffred !

eaudiffred, 2022-09-08:
Yeah, I think that sums it up.  On the first bullet I would say that once we get over 1000 units the injection molding option is what makes things potentially more cost effective.

There is a little bit of a gray area because units/time is a factor as well.  I think you are up to 3 printers (?) plus the 1 I have, so absolute best case is 4 units per 24hrs.  That might look more like 20-25/week.  I'm probably over thinking this, but if we have a steady demand then I think we could keep up, even if it means buying more printers.  If we have a surge, that would make things more difficult.  

If you think it would be useful, I could put together some forecasting tables to get a better idea of cost analysis per volume per time.  Maybe we aren't to the point where we need to worry about that yet, especially with part supply issues and scaleup plans still in the works.

Also, just a note that I did put a focus on finding a material that is recycled or recyclable and green.  That significantly limited our options, but I think both are important.

keenanjohnson, 2022-09-09:
I agree :)

muditprotect3d, 2022-10-05:
Is there anyway we could utilize the Duke Makerspace for this? I'd be more than happy to print a couple of enclosures and keep them as inventory and ship out as needed if that is convenient?

muditprotect3d, 2022-10-05:
I can print as much as I want for free as well if that's helpful

keenanjohnson, 2022-12-06:
I think we could for one off usecase @muditprotect3d , but it's not really a repeatable option for people to order parts from there as they are not an actual business or service.

I believe we completed the original goal with this task. We attempted to identify if there were any suppliers that were ideal for low volume 3d printing orders of the Frog Parts. There just don't seem to be any good options, so I'm going to close this issue.

closed 2022-12-06 by keenanjohnson