Ribbit-Network/ribbit-network-frog-hardware #132: Revise Frog Enclosure Mounts for Raspberry Pi CM4

keenanjohnson, 2022-07-26:
In #130 we identified that we need to go back to a Raspberry Pi CM4 version of the Frog Sensor.

This task is to revise the mechanical enclosure to fit the Raspberry Pi CM4 and Carrier Board as well as Antennae. 

An older version of the Onshape Assembly has the correct parts and geometry I believe, so we will just need to bring those further into the newer version of the assembly that has the better sensors at the top of the Frog.

V5 in OnShape: 
[image]

eaudiffred, 2022-07-26:
@keenanjohnson, A couple of questions for you if you don't mind.
1. I'm seeing different port locations on the Raspberry Pi board that I look up on the internet and the one that you have and used in V1.  Is the board you are using some kind of modified version?  The power port is in a much better location on the version shown above, but I can't seem to find that model.
2. In OnShape I took a stab at creating a branch, but I may have messed it up, sorry!  Should I continue working on the light Blue or Purple branch?

eaudiffred, 2022-07-26:
Oh, just saw your reply in Discord!
_"We actually used the Raspberry Pi CM4 with a different carrier board which is slightly different than the Raspberry Pi 4. We'll also need to find a spot for the external antennae."_

- What is the brand or model of the carrier board?  I'd like to find a dimension drawing.

- Which antenna needs to be used with the Pi?  The black on shown above, or the Adafruit one (https://www.adafruit.com/product/2308)?

keenanjohnson, 2022-07-26:
Yes great questions!

Here are the parts and links to datasheets!

* [Waveshare Mini Base Board (A) Designed for Raspberry Pi Compute Module 4](https://www.waveshare.com/product/raspberry-pi/boards-kits/compute-module-4-cat/cm4-io-base-a.htm)
* [CM4102000 Raspberry Pi Compute Module 4 Lite](https://www.newark.com/raspberry-pi/cm4102000/rpi-compute-module-4-lite-2gb/dp/86AH2093)
* [Raspberry Pi CM4 Antennae](https://datasheets.raspberrypi.com/cm4/cm4-antenna-kit-product-brief.pdf)

keenanjohnson, 2022-07-26:
One of the tradeoffs that I think needs to be made @eaudiffred is whether the antennae is going to live inside or outside the enclosure. The antennae is technically water resistant, however, it doesn't have an official IP rating or anything.

I had played around with it either way, but I'm not sure if you have opinions or thoughts @eaudiffred ?

External Ant. Example:

[image]

Internal Ant. Example:

[image]

eaudiffred, 2022-07-26:
@keenanjohnson, Thanks for the links! 

In my opinion, inside the enclosure is preferred for these reasons:

1. Less holes in the enclosure keeps external things out (rain, bugs, dust)
2. It presents a cleaner look
3. Over all footprint is reduced

However, if it compromises the range of the antenna, forces a much larger enclosure, or makes assembly overly difficult, I think having it stick out would be fine.  The cable and connections are protected either way.  

Do you know if the range is reduced if its housed inside?

keenanjohnson, 2022-07-26:
No the PETG of the enclosure shouldn't compromise the signal integrity as I understand it, so this is just a mechanical packaging issue :). So sounds like inside it is!

eaudiffred, 2022-07-26:
Cool.  Which branch should I work off of in OnShape?  I see you made one branched from an earlier bbgg edition.  Should I continue with that one?

keenanjohnson, 2022-07-26:
Oh I would work in the one you made! That has all the latest enclosure
stuff. Those other branches are are old from me, I just couldn't figure out
a way to delete them haha.

keenanjohnson, 2022-07-27:
No rush @eaudiffred , but just making sure you have everything you need here?

eaudiffred, 2022-07-28:
@keenanjohnson, yep.  Working on the antenna now.  I think that's the last change I need to make.

eaudiffred, 2022-07-28:
I think it's ready.  Had to extend the length of the enclosure and lid by 15mm.  I think that's an ok amount to keep it all enclosed.  And if I remember correctly we did shorten it when we switched to the BBGG board, so now it's back to the original.  

[image]

eaudiffred, 2022-07-28:
@keenanjohnson let me know if you see any glaring errors or parts that aren't fitting.  I have a head on my printer right now, but I'll try and get a base and lid going tomorrow.

keenanjohnson, 2022-07-28:
Sounds good @eaudiffred! We don't actually have to account for the USB cable at the top there any longer since we switched GPS units, but it looks like the ant. length is driving the height anyway. 

If we ditched the USB in the model, bent the ant. and moved it up and upside down, perhaps we could save some packaging space? Not sure if it is really worth it though. 

I'm thinking something like this:

[image]

This is definitely workable though and I'll try a print tonight and can fit check parts as well!

eaudiffred, 2022-07-28:
@keenanjohnson, sweet.  I actually had it modeled like that, but then changed it because of the usb connection, lol.  Bending it actually works pretty well.  I'll rework it and put in the bend.  Does anything need to be plugged into those top three ports?  That should save that 15mm too.

eaudiffred, 2022-07-28:
@keenanjohnson this might be a dumb question, but does it make a difference if the antenna is mounted upside down or right side up?

eaudiffred, 2022-07-28:
Here is option #1.
[image]

eaudiffred, 2022-07-28:
#2 would be flipping the antenna upside down.  Either way the antenna cable comes close to the entry/exit holes for the other cables, but I don't think it will cause a problem.

keenanjohnson, 2022-07-28:
Option 2 seems slightly worse to me from a packaging perspective because the part of the wifi ant. cable nearest the big plastic part of the ant. is the most stiff, making it hard to not poke the wifi cable with the power cable when you plug it into power.

[image]

For option 1, we can sort of route the cable around under the pi something like this:

[image]

eaudiffred, 2022-07-28:
Updated!
[image]

keenanjohnson, 2022-07-29:
Thanks @eaudiffred ! I'll do a test print soon.

keenanjohnson, 2022-08-04:
Hey @eaudiffred ! I had a chance to do the test print and put all the parts in for a fit check.

Overall, this seems pretty good and I like the way the cables will be routed:

[image]
[image]

Unfortunately, there is a misalignment on the height of the raspberry pi and the hole for the power cable as you can see below. Using a rule, my best estimate is that the hole probably needs to come down about 4 mm or so and things should be aligned.
[image]

keenanjohnson, 2022-08-04:
Once that's resolved, things should look nice.

[image]

eaudiffred, 2022-08-05:
@keenanjohnson, Excellent, glad the antenna fits.  I realize in the CAD assembly model the board is sitting on brass posts that are screwed into the plastic posts that we designed into the base.  That's why the hole for the power cord is off.  I'll bump down the hole and remove the brass from the assembly.  I should be able to knock it out tomorrow.

eaudiffred, 2022-08-05:
Those couple of changes have been made.  @keenanjohnson how did the antenna mounting go?  Was the hole correctly sized and could you tighten the nut without too much difficulty?

keenanjohnson, 2022-08-05:
Oh, I didn't realize the model had standoffs. I can also do those if you think they are important.

Oh yeah, the ant. mount seemed great! It's not exactly super easy to screw it in, but I think that is the ant. itself not the mount. I was thinking more about future improvements and perhaps in the future, maybe we could go to a clip-based system for the ant. instead? That's more of a future idea, perhaps the raspberry pi could be held in clips as well in the future, but perhaps that's not as robust as screws.

eaudiffred, 2022-08-05:
Removing the standoff is one less thing for people to have to install, so I think it makes sense to get rid of them.  You can install them in the one you have printed though to salvage that enclosure.  

I like the clip idea, getting rid of hardware is a good thing.  I think its possible with printing to design something in.  

When would you like to call V3 complete?  Should we continue to modify and experiment for now, or are we ready to crank out 30 of them to go along with the Pis that you have?

keenanjohnson, 2022-08-05:
Let's move the thought experiment of the clips to another issue and we can just pursue that in parallel as you have time to do the modeling. If you have time and it's ready to be part of V3 then that's great, but if not, no big deal.

eaudiffred, 2022-08-05:
Sounds good.  Are there other items that need to be completed for the launch of V3 or was the enclosure the last part?  Just trying to get a feel for what kind of timeline you are shooting for to finalize V3.

keenanjohnson, 2022-08-05:
Basically just this mechanical update and then updating the build instructions for v3 are left, so probably pretty quick. I hope to resolve v3 by mid-week maybe after I have time to do a test print / take new pictures for the instructions etc.

https://github.com/Ribbit-Network/ribbit-network-frog-sensor/milestone/3

keenanjohnson, 2022-08-23:
This is complete! I think we just need to update the mechanical files in the v3_docs branch on git.

keenanjohnson, 2022-09-01:
These files will be uploaded in #143

eaudiffred, 2022-09-22:
Completed and implemented as part of the V3 changes.

closed 2022-09-22 by eaudiffred