Ribbit-Network/ribbit-network-frog-hardware #133: Update BOM for Frog Sensor Version 3

keenanjohnson, 2022-07-26:


keenanjohnson, 2022-08-04:
https://github.com/Ribbit-Network/ribbit-network-frog-sensor/pull/137

keenanjohnson, 2022-08-04:
@pascaljoly I've updated the BOM for the V3 Frog Here. This is still a bit of a draft, but give it a review and I think it should have everything you need.

pascaljoly, 2022-08-04:
@keenanjohnson I did not see the BOM in your message. do you have a link for it?
thanks,
Pascal

keenanjohnson, 2022-08-04:
Oh it was in the PR I linked above: https://github.com/Ribbit-Network/ribbit-network-frog-sensor/pull/137, but you can also find it in the new V3 branch: https://github.com/Ribbit-Network/ribbit-network-frog-sensor/blob/frog_v3/hardware/v3/ribbit_network_frog_sensor_bom.csv

pascaljoly, 2022-08-04:
thanks. I trust you these are the correct parts. Is there a way to group
them in one public "shopping cart" link with adafruit?

keenanjohnson, 2022-08-22:
Merged #137 with the updated BOM

keenanjohnson, 2022-08-22:
I'm not sure if there is a way to do that @pascaljoly! That would be a great improvement if so possibly!

closed 2022-08-22 by keenanjohnson