Ribbit-Network/ribbit-network-frog-hardware #134: Revise Assembly Instructions for Frog Sensor V3

keenanjohnson, 2022-07-26:


fosteman, 2022-08-08:
Due to supply chain issues with Beagle and RPI, maybe we should also look into[ Orange PI (chinese alternative)](https://www.aliexpress.com/item/1005002907920819.html?spm=a2g0o.productlist.0.0.1dc844d3R9jtIP&algo_pvid=c77888de-c68d-4c87-a9bb-8335188a8ca7&algo_exp_id=c77888de-c68d-4c87-a9bb-8335188a8ca7-42&pdp_ext_f=%7B%22sku_id%22%3A%2212000022722815269%22%7D&pdp_npi=2%40dis%21CAD%2134.65%2134.65%21%21%21%21%21%402101d64d16599592240158844e2699%2112000022722815269%21sea&curPageLogUid=UDMUfr6O6psp)

[+ official website ](http://www.orangepi.org/)

eaudiffred, 2022-08-22:
@keenanjohnson, I have all of the parts listed in the V3 BOM.  I can take a shot at the assembly instructions for a full V3 sensor.  I need a little guidance on the hardware though.  I still have the V1 sensor up and running, and I know quite a bit has changed.  

- Do you have a wiring diagram, or just a sketch of what plugs into where for V3?  
- Do I need to download a new program to the sd card?
- Once it's assembled, I'll probably need some assistance connecting to wifi, but I can document that process as well.

keenanjohnson, 2022-08-22:
Hey @eaudiffred ! Wow yes that would be awesome if you wanted to jump in and help me document the process and getting all the pictures together for the build instructions! I'm going to go ahead and merge this PR for the V3 BOM and then we can add build instructions on top of that. 

* Hmmm I don't have a [proper wiring diagram really other than the instructions with pictures here.](https://github.com/Ribbit-Network/ribbit-network-frog-sensor/blob/main/hardware/v2/assembly-instructions/4-assembly.md). I will open another ticket to add that :) 
* You will need another program yes! The software instructions should be the same (other than selecting a different processor) as [the V2 software instructions here which link to the program.](https://github.com/Ribbit-Network/ribbit-network-frog-sensor/blob/main/hardware/v2/assembly-instructions/3-software.md)
* Indeed, let me know how I can help and where you get stuck!

keenanjohnson, 2022-08-23:
FYI @eaudiffred I have created a new branch in Git that we can use to update the V3 instructions and add the correct exported mechanical files. It's called [v3_docs](https://github.com/Ribbit-Network/ribbit-network-frog-sensor/tree/v3_docs) and you can see there is an empty folder for v3. 

I think we can start by just copying the docs from V2 and updating the pictures and instructions that have changed.

[image]

eaudiffred, 2022-08-23:
@keenanjohnson, thanks!  Thanks for sharing the link to the assembly instructions, I tent to get lost on GitHub pretty easily.  I'll work on getting updated pictures.  

How do I attach the new antenna to the carrier board?  That's the only step I can think of that I can't figure out from the V2 instructions.

eaudiffred, 2022-08-23:
Thanks!  I'll take a look this evening.  Mine does have the TPLink wifi dongle on it.  I'll take a picture though and confirm one way or another.

eaudiffred, 2022-08-24:
I'll work on the pictures and part files this weekend.  As you stated above, @keenanjohnson, I'll just copy the V2 files and then change anything that has been updated for V3.

eaudiffred, 2022-08-28:
I've taken the pictures for the V3 assembly.  I'm not exactly sure how to upload or copy files in Github though.  I need to upload the new STL files, and then copy the V2 assembly instructions and update.  Is this what a Pull Request does?  I'll try to find a youtube video explaining how to navigate github.  For now the pictures are in the google drive.

keenanjohnson, 2022-08-28:
@eaudiffred! That is excellent! Thank you so much! 

Unfortunately, Github is a bit complicated, but I'll try to help out! 

I've created a branch in github called "[v3_instructions](https://github.com/Ribbit-Network/ribbit-network-frog-sensor/tree/v3_instructions)".

We can add all the proper pictures and update the links in the document files to point to the proper pictures in that branch.

All the images can go in the [v3/assembly-instructions/images](https://github.com/Ribbit-Network/ribbit-network-frog-sensor/tree/v3_instructions/hardware/v3/assembly-instructions) folder and then we can update the links to the images in the v3/instructions folder. 

The mechanical files can be updated in the v3/hardware folder.

Let me know if you have any questions about how to add the files and edit with Github. I find using the Github desktop client to be easiest, but also if it's not in your interest to learn, please let me know and I can help out with that part of the updates.

Then we can create a pull request to review everything, I can add my pictures of the antennae, etc and merge it when we feel good.

eaudiffred, 2022-08-31:
@keenanjohnson I just sent a pull request with all of the new pictures and STL files for V3.  If this goes through successfully, then I think I have the hang of working in GitHub!  If it all looks ok on your end, then next I can start working on the assembly instructions updates.

keenanjohnson, 2022-09-01:
Yeah, it looks great! I left a comment in the pull request regarding next steps.

closed 2022-09-22 by keenanjohnson