Ribbit-Network/ribbit-network-frog-hardware #138: Explore Idea of Holding Raspberry Pi & Ant Via Clips Instead of Screws

keenanjohnson, 2022-08-05:
In #132 we discussed replacing the screws that hold the raspberry pi and the circular hole that holds the antennae with clips that are printed as part of the enclosure.

Not sure if that's practical or reliable, but an interesting idea to consider and try.

eaudiffred, 2022-08-05:
Added a branch in OnShape off of the final V3 enclosure to start experimenting with the clips

eaudiffred, 2022-08-09:
I have this prototype printing:
[image]
[image]

eaudiffred, 2022-08-10:
The clips are defiantly small, but didn't break off when I manually flexed them.  I have an antenna on order, hopefully it gets to me today or tomorrow and I can use the board in my V1 sensor to test fit.  Hopefully I can get to that this evening.  I'll share some pictures of the results.  

Totally up to you if you want to do a test as well.  

I also made some small modifications to the lid and enclosure base to remove the two screws.  I think the lid snaps in pretty well, and two less screws is a good thing.  It also allowed me to remove some material from the base.  
[image]
[image]

eaudiffred, 2022-08-10:
The board clips didn't work.  I made the dimensions too exact.  Need to add in some clearance for the post height, clip height and clip width.

keenanjohnson, 2022-08-10:
The new lip on the enclosure looks great. Thanks so much for tackling this!

eaudiffred, 2022-08-12:
I received the antenna yesterday and tried it out.  The base clip was too tight and snapped.  I'll tweak both sets and try another test print.

eaudiffred, 2022-08-22:
This isn't going as smoothly as I hoped.  I'll need some more time.  I think it would be best to launch V3 as is concerning the enclosure.

keenanjohnson, 2022-08-22:
Makes sense as this is a pretty difficult challenge it seems to me! It's easy to do another revision after once we figure it out :)

eaudiffred, 2022-09-02:
Here is my latest idea for the clip design.  
[image]
In the previous versions the vertical/post style clips kept snapping.  In this version, only the zig-zag shape clips will flex.

keenanjohnson, 2022-09-02:
Awesome! Do you need any help doing a test print @eaudiffred ?

eaudiffred, 2022-09-02:
I can get a test printed this weekend.  But obviously feel free to print one as well!  
How is the new printer working out?  Big difference?

keenanjohnson, 2022-09-07:
The new printer is much nicer. The auto bed leveling sensor saves a ton of time it seems and I've zero failed prints so far (only printed a few test things though) which is super nice.

eaudiffred, 2022-09-08:
Much better results with the clips shows in the CAD model above.  Some tweaking to do on the upper antenna clip and tightening up the board clips.

eaudiffred, 2022-09-26:
I'm going to pause on this until we determine if we need to move to V4 of the sensor to support new hardware.