Ribbit-Network/ribbit-network-frog-hardware #156: time scale display on all graphs / heroku

pascaljoly, 2022-09-28:
when I look at the graphs on https://ribbit-network.herokuapp.com/ the time scale is only displayed in the last graph (Humidity). Can we also add it to the other graphs?

jklmaynard, 2022-09-29:
I suppose it's better to put this on the ticket than in Discord.

I can't test it, because my local server can't SSH into the online data.  If there's a guide for implementing the DataVisualization csvs, let me know.  Otherwise, I think the fix for the time scale is to remove `shared_xaxes=true` from line 201 of the app.py file.

Here's the documentation from Plotly on that: https://plotly.com/python/subplots/#subplots-with-shared-xaxes

@pascaljoly @spestana

keenanjohnson, 2022-09-29:
Hey @pascaljoly ! I think this issue should actually be in the dashboard project, no the Frog sensor project. I can transfer it now.