Ribbit-Network/ribbit-network-frog-hardware #158: ESP32 Frog: Find & Integrate Sensor Libraries into ESP-IDF Framework

keenanjohnson, 2022-10-01:
As part of the effort to create an EPS32 based Golioth Frog Sensor #126 , we need to find and integrate drivers for the Frog Sensors into the ESP-IDF framework stared in this branch: (https://github.com/Ribbit-Network/ribbit-network-frog-sensor/tree/esp32_golioth)

**Sensors**
* SCD-30 - NDIR CO2 Temperature and Humidity Sensor - https://www.adafruit.com/product/4867
* DPS310 Precision Barometric Pressure / Altitude Sensor - https://www.adafruit.com/product/4494
* GPS PA1010D - https://www.adafruit.com/product/4415

beriberikix, 2022-10-01:
There's a common set of sensor drivers [UncleRus/esp-idf-lib](https://github.com/UncleRus/esp-idf-lib) which has support for the SCD-30.

`esp-idf-lib` currently doesn't support the DPS310 but I just filed UncleRus/esp-idf-lib#456. It's a simple I2C so hopefully we can get some support there.

GPS in general is usually standard NMEA, so something like https://github.com/igrr/libnmea-esp32 should work.

I don't have the sensors on hand but I'm going to order them and see if I can help with the validation :)

keenanjohnson, 2022-10-04:
Thanks @beriberikix for pointing out those libraries! Hopefully it shouldn't be too difficult to add those into my branch I have! I will try to work on it soon as I have all the sensors here I can validate with.

keenanjohnson, 2022-10-08:
I have linked the [UncleRus/esp-idf-lib](https://github.com/UncleRus/esp-idf-lib) library into my golioth project and imported the scd30 module. I should be able to verify the functionality shortly.

https://github.com/Ribbit-Network/ribbit-network-frog-sensor/actions/runs/3208759367

keenanjohnson, 2022-10-18:
I've created three separate Github issues for each of our sensors to track and divide this work as part of the milestone: https://github.com/Ribbit-Network/ribbit-network-frog-sensor/milestone/4

#173 
#174 
#175

closed 2022-10-18 by keenanjohnson