Ribbit-Network/ribbit-network-frog-hardware #163: Enclosure Updates for ESP32 Based Frog

keenanjohnson, 2022-10-06:
Hey @eaudiffred !

As we discussed in our team call earlier, I have fairly high confidence in the new processor that we would use for the microcontroller based version of the Frog.

Here is the actual part and link to adafruit: [Adafruit ESP32-S3 Feather with 4MB Flash 2MB PSRAM - STEMMA QT / Qwiic](https://www.adafruit.com/product/5477)

[image]

**Implementation Notes**

The power connector is a USB Type connector again shown on the left hand side of the image below. I imagine we would want to orient the board such that the power connector faces down like our previous designs

[image]

One thing that is a bit annoying is that USBC Power Adapters tend to be fairly large and not designed for outdoor applications, so I'm wondering if it might be worth designing around a [USBC to Barrel Jack to USB C Adapter like this one](https://www.adafruit.com/product/4536)?

Electrically this would be fine, but physically it will be perhaps mechanically wasteful as that adapter is almost the same length as the whole ESP board haha.

Would love any thoughts or opinions there :)

[image]

eaudiffred, 2022-10-07:
Thanks @keenanjohnson.  One initial question about the power supply.  Are there any USB-C power supply units that will work?  It looks like it's 3V and I think a max on 150mA.  I can look around for something so that we don't have to use an adapter.  If you know it doesn't exist, just let me know and I'll tie in the adapter.

eaudiffred, 2022-10-07:
Just posting this here for reference.  
[image]

eaudiffred, 2022-10-08:
I believe this is a STEP file of the board, but I can't seem to figure out how to download it.  It's my lack of GitHub knowledge I think.
https://github.com/UnexpectedMaker/esp32s3/blob/main/3d%20models/FeatherS3_P4.step

eaudiffred, 2022-10-08:
Got it!  Remembered the Desktop App.  The board is not in the CAD imports folder.

keenanjohnson, 2022-10-08:
@eaudiffred ! Any standard 5V Usb C Power Supply should work (where did you see the 3V number)? I've confirmed that here :)

eaudiffred, 2022-10-10:
@keenanjohnson, ok cool.  That eliminates the need for the adapter, correct?  
I misread the data sheet.  I thought it was limited to 3V, but that is incorrect.

keenanjohnson, 2022-10-10:
There are adapters, but the USBC plug I've read does less well outdoors overtime than a barrel jack, but perhaps that's not really a problem worth solving for the moment until we are sure we have an issue.

eaudiffred, 2022-10-10:
I've started to work on the enclosure based on the ESP32 and we will be able to reduce the size significantly.  Here is an initial idea for the model.

[image]
[image]

keenanjohnson, 2022-10-10:
Wow this would be excellent if we could take the size down significantly! 

Looking at your renderings, I'm a little bit suspect that there is enough room there for the USBC cable. Unfortunately, the power connector on the cable end is usually quite long. We might need to include that in the model to check out if there is enough space?

This thing [from a common USBC suply](https://www.adafruit.com/product/4298)

[image]

eaudiffred, 2022-10-10:
@keenanjohnson, good point.  I went ahead and make a model for the USBC supply.  It does cause issues.  We may be able to reduce the width of the enclosure by taking out space between the clips (red circles).  Then rotating the whole board to almost vertical and lengthening the base.  

I think I can squish the depth a little too.  The width and depth adjustments would make the shroud and head incompatible between versions, but I don't see that as a huge problem right now.  The amount of material and time that can be saved seems to be worth the changes.

[image]

keenanjohnson, 2022-10-10:
Yeah, I wouldn't worry about head capability between versions at the moment. 

One thing we could consider is a right-angle USB C adapter? Perhaps this would help us with the cable routing out of the side? 

https://www.adafruit.com/product/4432

[image]

eaudiffred, 2022-10-10:
Yeah, I was thinking about that and weighing pros and cons of one additional part on the BOM.  I'll model that up too and see what configuration seems to save the most material.

eaudiffred, 2022-10-12:
I was able to shrink the depth and width a decent amount.  We are able to save quite a bit of material by using this smaller board.  
 
[image]

[image]

eaudiffred, 2022-10-12:
@muditprotect3d, now would be a great time to introduce any ideas regarding the hanging mechanisms!  While we are remodeling for the ESP32 we will have lots of opportunity to prototype.  Also, any ideas concerning the enclosure in general is more than welcome!

muditprotect3d, 2022-10-12:
@eaudiffred sounds good! I will add some to the thread right now

muditprotect3d, 2022-10-12:
[image]

muditprotect3d, 2022-10-12:
[image]
This is another one, basically similar to the last one except would span the entire width of the back of the enclosure

muditprotect3d, 2022-10-12:
[image]
This one would be a variable one where you could slide the two ends across the channel to fit across any component of varying length

muditprotect3d, 2022-10-12:
I created it on the shroud but the ideal location would be on the back of the enclosure

muditprotect3d, 2022-10-12:
[image]
These are more simple hooks, but im working on a design right now to make the bottom hook slideable through a channel

muditprotect3d, 2022-10-12:
Im working on another design this week, and I had some ideas about the enclosure as well I can send in here by end of week

eaudiffred, 2022-10-13:
@muditprotect3d Awesome work!  Thanks for sharing.  I realize we should document this design work in a separate GitHub issue just to keep projects organized.  I'll open a GitHub issue for Hanging Mechanism Designs, and we can keep developing these ideas in there.  I'll tag you and reference the posts above.  Then we can use this issue for enclosure design related to the ESP32 change.

muditprotect3d, 2022-10-17:
Reading this thread again, have we considered/thought about using the space tradeoff to include an external battery instead of a power cable? ( Instead of reducing the size of the enclosure, we could use the extra space due to a smaller processor to include an external battery/power source)?

keenanjohnson, 2022-10-18:
Great question @muditprotect3d! In the future once we create a solar powered version of the frog, we will need a battery, but right now even with the microcontroller, we will need to have an external power source to power things. The power management for a battery system is a bit complex and I think needs to be designed first before we could scope modifying an enclosure around that.

eaudiffred, 2022-10-21:
I have some prototypes printed and was able to test fit with the main electrical components.  
CAD files can be found [here](https://cad.onshape.com/documents/b3e6eeabf50d585d20f25fc6/w/da7f4373da69947f3d05d9c4/e/634db96a0c5d4c7b0ff00c45).

Here are some pictures comparing V3 and V4:
[image]
[image]
[image]

Here are some pictures showing the electrical component fit.  
[image]
[image]
[image]

eaudiffred, 2022-10-21:
The biggest problem remaining is figuring out the clip design for the ESP32 board.  I still have not gotten a reliable design that has enough flex for repeated use.  I keep snapping the clips when inserting the board on the 1st or second try.  Let me know if anyone has ideas!  The goal here is just to reduce the amount of hardware (screws in this case) needed for the sensor.    

[image]

keenanjohnson, 2022-10-21:
Excellent work @eaudiffred ! Were you able to try plugging in the USB C cable while it's in the enclosure? I'm wondering if there will be any problems with the weight of the hanging cable tilting the hanging sensor and allowing more water into the shields at a high angle as the sensor is getting so small and light now. 

Perhaps @muditprotect3d might have some suggestions on the mechanical side for the clips?

eaudiffred, 2022-10-22:
@keenanjohnson, excellent point on the cable weight.  I just hung the sensor and plugged it in.  It did tilt slightly.  Enough that its worth moving the power cable entry point.

eaudiffred, 2022-10-22:
I went with a slightly different approach and flipped the ESP32.  
[image]
[image]
[image]

muditprotect3d, 2022-10-22:
@keenanjohnson @eaudiffred could we do something with sliding it in and having it be an interference fit? This way the plastic material could hold the circuit in place similar to how the screw holes work? It would essentially be like a sliding compartment that we put the controller into to keep it from moving around, but wouldn't require any clips to flex back and forth

eaudiffred, 2022-10-22:
@muditprotect3d, Yes that's a great idea!  I think that will actually keep the board more secure and be easier to print.

muditprotect3d, 2022-10-23:
@eaudiffred Awesome! Would it be useful if I tried to make some CAD models of what I'm thinking here?
I'm also working on the hanging mechanism incorporated into the larger device. I have two tests this week so am a little swamped but will get that done by midweek!

eaudiffred, 2022-10-23:
@muditprotect3d, yeah that would be great!  Here is the most [recent model](https://cad.onshape.com/documents/b3e6eeabf50d585d20f25fc6/w/da7f4373da69947f3d05d9c4/e/c221edb50cd05a98d22970e2) I have been working on.   

No rush, get those tests out of the way before worrying about this.  Let me know if you run into any blockers or need help on anything.  Good luck on the tests!

muditprotect3d, 2022-11-21:
[image]
putting this here but here is a mockup of the sliding rails for the feather I was talking about

eaudiffred, 2022-11-21:
Just copying this comment here (from Discord) for reference.
One thing to keep in mind is that the board will need to slide into place, correct?  So there needs to be enough space in the enclosure to drop the board down, and then slide it over.  Hopefully my thinking makes sense.  Let me know if I need to explain via picture.

keenanjohnson, 2022-11-21:
Great work collaborating all! Is it worth it creating a separate issue for this sliding feature or are you all ok collaborating here in this issue?

eaudiffred, 2022-11-23:
@keenanjohnson I think we can finish up the sliding feature in this issue and then call it complete.  @muditprotect3d could you share the link to the file you've been editing?

muditprotect3d, 2022-11-24:
Yes sending now!

muditprotect3d, 2022-11-24:
https://cad.onshape.com/documents/3d928932fd41fdc8e9fbaac0/w/87262cd35788e1173a2d535a/e/c180c5717ccf65cb0e58b6c0?renderMode=0&uiState=637f02b3c69bdb43ca62accc

muditprotect3d, 2022-11-24:
I'll make the edits you had suggested by the end of the week and have a final CAD prototype ready to print by EOW. Sorry about the delay on my end with this.

eaudiffred, 2022-11-28:
@muditprotect3d Looks good!  I copied your work over to the main file and added some fillets and chamfers.  I'll try and get a test print out this week as well.  
[image]

muditprotect3d, 2022-11-28:
Awesome! I think the tolerancing may have to be increased but it will be good to get some prints out and see

keenanjohnson, 2022-11-28:
Yes let me know if you have anything that is ready for a test print as well.

keenanjohnson, 2022-11-29:
@muditprotect3d I printed the file you sent in Discord and in general, I think it looks great, though there are some issues to sort out. Thanks for your hard work and @eaudiffred for supporting as always!

**Board Doesn't Fit in Slot**

The biggest issue is that the board is too wide to fit in the slots as provided. The 3D model of the board in Onshape must be wrong or the tolerances are too tight. 

Additionally I noticed there isn't too much space to wiggle the board down in line with the slot, even if the slot was the correct width. Perhaps the whole enclosure would need to be wider?

[image]

[image]

**Cable Bend Radius**
I'm pretty worried about bending the USB cable 180 degrees like this. The cables are generally pretty stiff and I noticed it really puts a lot of tensions on the connector in the USB housing. I'm wondering if it makes more sense to go back to the esp 32 being vertically mounted? If not, we may need a feature to provide strain relief for the cable so it is not transferred to the connector on the board which is not very strong I think.

[image]

**Board Dimensions**

I've include the measurements I took of my board for validation of the dinesnsions from Adafruit's drawings.

[image]

[image]

eaudiffred, 2022-11-29:
Great feedback @keenanjohnson, thanks!  I also printed one last night, but I tweaked the dimensions ahead of time.  I was able to get the board to slide in, but it was still a little too tight.  I think 22.7mm, the exact width of the board will work well.  
[image]

eaudiffred, 2022-11-29:
I also think that the slides can be shorter.  Something like this will provide enough support.  The slides are 20mm long in this picture.  
[image]

eaudiffred, 2022-11-29:
@muditprotect3d great work on the design!  I like how this functions and secures the board.

eaudiffred, 2022-11-29:
@keenanjohnson, I'm wondering if we should add the 90deg adapter.  I hate to add another piece to the BOM, but I like how much time and material we've been able to save.  

I'm thinking of two other options right now (the 90deg spacer being a 3rd option):
[image]
[image]

muditprotect3d, 2022-11-29:
Hey @eaudiffred @keenanjohnson thank you for the feedback! I didn't have a feather with me when making the slots so I was using the measurements on onshape and kind of figured they would be off a little but this is super helpful feedback! @eaudiffred are the edits you mentioned on the main file? I can go ahead and get another test print as well with the 20mm slots

muditprotect3d, 2022-11-29:
> Great feedback @keenanjohnson, thanks! I also printed one last night, but I tweaked the dimensions ahead of time. I was able to get the board to slide in, but it was still a little too tight. I think 22.7mm, the exact width of the board will work well.
> [image]
we can increase tolerance in the slide then if it was too tight fitting in?

eaudiffred, 2022-11-29:
@muditprotect3d, Yes.  I put the edits in the main file.

muditprotect3d, 2022-11-29:
@eaudiffred sounds good! Sorry about the dimensioning issues there. I'll also get a test print going for today

eaudiffred, 2022-11-29:
@muditprotect3d, no worries!  That's why we test print!  Definitely an iterative process.

muditprotect3d, 2022-11-29:
I think that could greatly reduce the strain in the wire -- if we want to make it slightly simple in terms of design we could make it tolerances clips against the side of the wall that the wire can be pushed into?

muditprotect3d, 2022-11-29:
[image]
@eaudiffred in terms of the modular hooks I was thinking of something like this essentially with a similar hook mechanism as the shroud and then have different types of hooks available for the user to print? Not sure if that's what you had in mind but we can talk more during our meeting tomorrow as well!

keenanjohnson, 2022-11-29:
Do you have any bandwith to take a shot at design of those strain relief clips @muditprotect3d and I could do a test print and assembly?

As for the hanging hooks, not to be annoying, but let's try to keep that discussion separate in #171

muditprotect3d, 2022-11-29:
@keenanjohnson sorry about that! I have a midterm tomorrow but I can take a shot at those clips right after that if that works?

muditprotect3d, 2022-11-29:
@keenanjohnson I think @eaudiffred was doing some work on changing the enclosure design for this issue here as well: https://cad.onshape.com/documents/b3e6eeabf50d585d20f25fc6/w/da7f4373da69947f3d05d9c4/e/634db96a0c5d4c7b0ff00c45

muditprotect3d, 2022-11-29:
[image]
I have a test print in process right now for this enclosure structure

eaudiffred, 2022-11-29:
@keenanjohnson, @muditprotect3d, I've been messing around with the base this morning trying different shapes.  I have some time to build in the strain relief.  I can get it done in the next few hours and have it ready for a test print.  I'll report back when done.

[31 more replies]

closed 2023-01-26 by keenanjohnson