Ribbit-Network/ribbit-network-frog-hardware #172: Store Wifi Credentials After OTA Update

keenanjohnson, 2022-10-18:
Our device will need to store the wifi credentials in a way that they can be saved and reapplied after a device firmware update.

keenanjohnson, 2022-10-18:
@akhilgupta1093 are you still up for tackling this? 

I just made this post in the golioth forums to see if they had any examples or best practices to look at for this: https://forum.golioth.io/t/store-wifi-credentials-after-ota-update/110

akhilgupta1093, 2022-10-18:
Hi @keenanjohnson -- the [response](https://forum.golioth.io/t/store-wifi-credentials-after-ota-update/110) provided by Jonathan seems to make sense to me. 
The code that you implemented calls the [NVS API](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html#introduction) to store the provided wifi credentials outside of the firmware. NVS stores other device-specific credentials, like the PSK ID. It seems like OTA updates won't interfere with these stored values, then. What do you think? If you do an OTA update, do you see this behavior?

keenanjohnson, 2022-11-27:
As part of https://github.com/Ribbit-Network/ribbit-network-frog-sensor/pull/184, I have validated that this functions.

closed 2022-11-27 by keenanjohnson