Ribbit-Network/ribbit-network-frog-hardware #174: ESP32: Integrate GPS Library into ESP-IDF Framework

keenanjohnson, 2022-10-18:
As part of the effort to create an EPS32 based Golioth Frog Sensor https://github.com/Ribbit-Network/ribbit-network-frog-sensor/issues/126 , we need to find and integrate drivers for the Frog Sensors into the ESP-IDF framework stared in this branch: (https://github.com/Ribbit-Network/ribbit-network-frog-sensor/tree/esp32_golioth)

**Sensor**
GPS PA1010D - https://www.adafruit.com/product/4415

keenanjohnson, 2022-10-18:
@beriberikix suggested in #158 that this https://github.com/jacketizer/libnmea might work for our GPS.

However, I'm not quite sure since our GPS uses I2C as the physical interface and I think the library above is linked only for serial / NMEA strings, but I might be wrong.

keenanjohnson, 2022-10-18:
There is an arduino supported library for the I2C interface, so it may be possible to port that source into ESP-IDF?

https://learn.adafruit.com/adafruit-mini-gps-pa1010d-module/arduino

beriberikix, 2022-10-18:
Ya, that's correct! If you take a look at https://github.com/igrr/libnmea-esp32 you can see how it's modified to work with the esp-idf.  So the steps could be:
* Include `libnmea` into the project like @igrr's example
* Adapt to use [i2c](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2c.html) instead of [uart](https://github.com/igrr/libnmea-esp32/blob/main/example/main/nmea_example_main.c#L34)
* Update to ensure you're grabbing the right NMEA data

beriberikix, 2022-10-18:
I also created https://github.com/UncleRus/esp-idf-lib/issues/460.

akhilgupta1093, 2022-10-21:
Thanks @beriberikix, still learning all of this, but it seems like step 2 above is the uncertain one :). I'll play around with the code, but definitely will be tracking the issue under esp-idf-lib.

keenanjohnson, 2022-12-06:
@akhilgupta1093 let me know when this is at a stopping place for now and I can move the esp32 esp-idf branch into the new repo at https://github.com/Ribbit-Network/ribbit-network-frog-software