Ribbit-Network/ribbit-network-frog-hardware #175: ESP32: Integrate SCD30 Sensor Library

keenanjohnson, 2022-10-18:
As part of the effort to create an EPS32 based Golioth Frog Sensor https://github.com/Ribbit-Network/ribbit-network-frog-sensor/issues/126 , we need to find and integrate drivers for the Frog Sensors into the ESP-IDF framework stared in this branch: (https://github.com/Ribbit-Network/ribbit-network-frog-sensor/tree/esp32_golioth)

**Sensors**
SCD-30 - NDIR CO2 Temperature and Humidity Sensor - https://www.adafruit.com/product/4867

There's a common set of sensor drivers [UncleRus/esp-idf-lib](https://github.com/UncleRus/esp-idf-lib) which has support for the SCD-30.

keenanjohnson, 2022-10-18:
There is an example project here showing how to integrate the SCD-30 library

https://github.com/UncleRus/esp-idf-lib/tree/master/examples/scd30/default

keenanjohnson, 2022-10-18:
The basic proof of concept code is now complete for reading data from the SCD30 sensor. Will validate with hardware shortly.

keenanjohnson, 2022-10-18:
Build log is here: https://github.com/Ribbit-Network/ribbit-network-frog-sensor/actions/runs/3271380897/jobs/5381101675

keenanjohnson, 2022-10-18:
I'm attempting to test this with the hardware and discovered that the Adafruit dev board we are using has a switchable power source for the I2C connector that gets disabled when I flash the board.

[image]

I'm trying to set the pin controlling that power source to high, but I can't seem to make it work. Perhaps you have some insight into what I might be doing incorrectly @beriberikix?

https://github.com/Ribbit-Network/ribbit-network-frog-sensor/blob/esp32_golioth/software_esp32/main/app_main.c#L178-L181

beriberikix, 2022-10-18:
Oooh, I didn't know they added that. Very useful for future battery readings as well as low-power mode. I think there's a bit of initialization missing, which is probably masked by scd30 library. Fortunately there's also a library for the battery gauge that will be useful for future features, not just enabling: https://esp-idf-lib.readthedocs.io/en/latest/groups/lc709203f.html

I would suggest breaking this up a bit further - just getting each one of the I2C devices working in a standalone example, integrate them one at a time and _then_ add Golioth. 

If that fails, or just because, hop over to our [forum](https://forum.golioth.io/) :)

keenanjohnson, 2022-10-24:
I tried flashing a few more times and I actually believe this is an issue with the debugger not resetting the esp32 properly after the flash. When I manually power cycle after the flash, this works as expected. Will need to investigate more.

keenanjohnson, 2022-10-24:
> Ah yes, that's one annoying thing about the ESP32 Feathers - you have to
> put it in "programming mode"...sometimes. Have you put it into bootloader
> mode
> <
> https://learn.adafruit.com/adafruit-esp32-s2-feather/factory-reset#step-2-enter-rom-bootloader-mode-3106832

Yes, that’s the way I get it to flash most of the time unfortunately. Is
that why I have to manually power cycle after to get it out of programming
mode at the boot loader level?

Any work around a or documentation that better explains the interactions
between the boot loader, esp-idf and everything I should read?

keenanjohnson, 2022-10-25:
Nevermind, after reading the adafruit docs on this board, the behavior is quite clear. 

I have a consistent process to flash using the reset button, however, I believe I have two separate problems. 

1. The scd30 code from the example in doesn't seem to function. I have filed an issue to get to the bottom of that here: https://github.com/UncleRus/esp-idf-lib/issues/464
2. I'm having some problems setting the golioth nvm wifi credentials using the terminal program. 

I get this error from the terminal and no commands seem to produce any change or output in the terminal.

```
Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.
Press Enter or Ctrl+C will terminate the console environment.

Your terminal application does not support escape sequences.
Line editing and history features are disabled.
On Windows, try using Putty instead.
esp32> W (1646) golioth_example: WiFi and golioth credentials are not set
W (1646) golioth_example: Use the shell settings commands to set them, then restart
```

keenanjohnson, 2022-11-27:
I've completed this task.

I have created an esp-idf project that integrates the SCD30 library and the latest golioth firmware from https://github.com/golioth/golioth-firmware-sdk

I have validated that the code does talk to the SCD30, connects to golioth, etc

See PR for code details: https://github.com/Ribbit-Network/ribbit-network-frog-sensor/pull/184

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420a4b6a
0x420a4b6a: esp_pm_impl_waiti at /Users/keenan/esp/esp-idf/components/esp_pm/pm_impl.c:839

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1650
load:0x403c9700,len:0xb9c
load:0x403cc700,len:0x2f98
entry 0x403c994c
I (24) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (24) boot: compile time 12:56:53
I (25) boot: chip revision: 0
I (26) boot.esp32s3: Boot SPI Speed : 80MHz
I (31) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 4MB
I (40) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (64) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (72) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  3 factory          factory app      00 00 00010000 00100000
I (87) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (94) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (101) boot: End of partition table
I (106) boot: Defaulting to factory image
I (110) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=22f1ch (143132) map
I (145) esp_image: segment 1: paddr=00032f44 vaddr=3fc97180 size=0426ch ( 17004) load
I (149) esp_image: segment 2: paddr=000371b8 vaddr=40374000 size=08e60h ( 36448) load
I (158) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=a8d10h (691472) map
I (283) esp_image: segment 4: paddr=000e8d38 vaddr=4037ce60 size=0a31ch ( 41756) load
I (292) esp_image: segment 5: paddr=000f305c vaddr=50000000 size=00010h (    16) load
I (300) boot: Loaded app from partition at offset 0x10000
I (301) boot: Disabling RNG early entropy source...
I (302) cpu_start: Pro cpu up.
I (305) cpu_start: Starting app cpu, entry point is 0x40375370
0x40375370: call_start_cpu1 at /Users/keenan/esp/esp-idf/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (320) cpu_start: Pro cpu start user code
I (320) cpu_start: cpu freq: 160000000
I (321) cpu_start: Application information:
I (321) cpu_start: Project name:     golioth-esp-idf-external-app
I (321) cpu_start: App version:      hw_v2-93-g1f33572-dirty
I (321) cpu_start: Compile time:     Nov 27 2022 12:56:47
I (322) cpu_start: ELF file SHA256:  aeb0a1ec7b541b07...
I (322) cpu_start: ESP-IDF:          v4.4.2
I (322) heap_init: Initializing. RAM available for dynamic allocation:
I (322) heap_init: At 3FCA2688 len 0003D978 (246 KiB): D/IRAM
I (323) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (323) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (323) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (324) spi_flash: detected chip: generic
I (325) spi_flash: flash io: dio
I (325) sleep: Configure to isolate all GPIO pins in sleep state
I (326) sleep: Enable automatic switching of GPIO sleep configuration
I (327) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
[... 42 lines cut ...]
W (1749) wifi:<ba-add>idx:0 (ifx:0, 28:bd:89:de:66:df), tid:6, ssn:2, winSize:64
I (1829) wifi:BcnInt:102400, DTIM:2
W (1959) wifi:<ba-add>idx:1 (ifx:0, 28:bd:89:de:66:df), tid:0, ssn:0, winSize:64
I (2649) esp_netif_handlers: sta ip: 192.168.86.92, mask: 255.255.255.0, gw: 192.168.86.1
I (2649) example_wifi: WiFi Connected. Got IP:192.168.86.92
I (2649) example_wifi: Connected to AP SSID: Bill Wi the Science Fi
I (2649) golioth_mbox: Mbox created, bufsize: 1144, num_items: 10, item_size: 104
I (2659) golioth_example: SCD30 Firmware Version: 3.2
I (2659) golioth_example: Starting continuous measurement
I (2699) golioth_coap_client: Start CoAP session with host: coaps://coap.golioth.io
I (2709) libcoap: Setting PSK key

I (2709) golioth_coap_client: Entering CoAP I/O loop
I (2959) golioth_example: Golioth client connected
I (2959) golioth_coap_client: Golioth CoAP client connected
I (4669) golioth_example: CO2: 838 ppm
I (4669) golioth_example: Temperature: 24.89 °C
I (4669) golioth_example: Humidity: 35.14 %
I (6679) golioth_example: CO2: 942 ppm
I (6679) golioth_example: Temperature: 24.82 °C
I (6679) golioth_example: Humidity: 35.24 %
I (8689) golioth_example: CO2: 967 ppm
I (8689) golioth_example: Temperature: 24.80 °C
I (10699) golioth_example: CO2: 974 ppm24 %
```

closed 2022-11-27 by keenanjohnson