Ribbit-Network/ribbit-network-frog-hardware #181: Beaglebone not showing up as option on balena network

eliasfallon, 2022-11-24:
I freely admit I'm very behind. I know most of Ribbit-network is working on v3 of the hardware now. I bought all my parts (v2) almost a year ago, and am just finally getting around to putting everything together now, and trying to assemble and install. But when I get here: https://hub.balena.io/organizations/g_keenan_johnson1/fleets/ribbit-network it is only showing Raspberry Pi options, no Beaglebone options. When I check the code: https://github.com/Ribbit-Network/ribbit-network-frog-sensor/blob/main/software/balena.yml, it seems to be showing that BeagleBone Green should still be on the list, but it isn't visible at Balena. Any suggestions or help are greatly appreciated.

eliasfallon, 2022-11-24:
Happy Thanksgiving everybody! (apologies for posting an issue on the holiday, first chance I've gotten to actually put my hardware together since I got the 3D parts printed). I did try searching both github and discord for similar issues before and didn't find anything, my apologies if I missed it.

keenanjohnson, 2022-11-25:
Hey @eliasfallon ! No worries at all, I think you are the first one to encounter this. Let me look into it and get you a solution shortly.

keenanjohnson, 2022-11-27:
@eliasfallon I have resolved this. 

I'm not sure why it disappeared from the list, but to make things more clear and for some other software reasons, I've spit our balena fleet into different fleets per version.

There is now a new V2 Fleet here: https://hub.balena.io/organizations/g_keenan_johnson1/fleets/ribbit-network-frog-v2

You should be able to add your device to that fleet now :) 

I've also updated the build instruction links for V2, but let me know if this works and if you run into any other issues.

keenanjohnson, 2022-11-27:
I'll leave this issue open until I hear confirmation from you that you that this is resolved.

eliasfallon, 2022-11-27:
Hey @keenanjohnson - With the 'v2' version of the fleet, I was able to get the image setup, downloaded and flashed to my SDcard. So that part worked, still in the process of getting the sensor fully setup, but I wanted to let you know that worked, and I think I know why it wasn't showing up before. From the docs here (all the way at the bottom): [https://hub.balena.io/what-is-balenahub](https://hub.balena.io/what-is-balenahub )- it says:

> Why aren’t all the devices I support in my `balena.yml` file showing on my fleet?
> You can build an application and associated balena.yml file that supports multiple device types of all different architectures, but the fleet on the balenaHub side can only currently support a single architecture at any one time. This is why you're seeing a cut down list - the platform is taking the list of supported devices you provide in balena.yml and filtering it to show only those device types compatible with the same architecture as the default device type selected for that fleet. This is why you'll see fleets published with the architecture in the fleet name - they're publishing one fleet per architecture and pushing the same code to each.
> We do plan to address this in the future, but it is a limitation today that requires that you make multiple fleets to target each architecture that you would like to support individually. Alternatively, you can list your submission as an app, and it will show all the supported device types and simply ask the user to choose the one they want to use when they create a new fleet and deploy the app. The fleet will still only allow devices of a single architecture at any one time however.

keenanjohnson, 2022-11-27:
Thanks @eliasfallon ! Please reach out again if you have any other issues :)

Interesting. It seems like that behavior of Balena changed since the last time I read the docs. Anyway, I guess it's good validation for splitting our fleets :)

closed 2022-11-27 by keenanjohnson