Ribbit-Network/ribbit-network-frog-hardware #199: suggestions to improve v4 software install

pascaljoly, 2023-01-26:
Hello,
I wanted to suggested a few items for the v4 software install: https://github.com/Ribbit-Network/ribbit-network-frog-hardware/blob/main/v4/assembly-instructions/3-software.md
1. Connect the esp32 board to your computer via a USB C Cable: make sure your cable does support data. many USB cable are for charging only. example of a cable you can use: https://www.amazon.com/dp/B0794M53HQ?ref=ppx_yo2ov_dt_b_product_details&th=1 -- one way to verify your cable is of the correct type: as soon as you plug in the ESP32 your computer (PC or Mac) should detect a new hardware and prompt you for installing drivers if missing. another hint that you have the correct cable: the JTAG serial port should be listed when you hit CONNECT.
2. Hold the "boot" button and press the "reset" button (shown below). This puts the esp32 into a flashing mode: rephrase to: Hold the "boot" button (red square) then press the "reset" button (blue square) as shown below. This puts the esp32 into a flashing mode. the lights of your ESP32 should turn off.

keenanjohnson, 2023-01-26:
Great suggestions @pascaljoly !

eaudiffred, 2023-04-02:
Updated

eaudiffred, 2023-04-02:
PR #213

closed 2023-04-02 by eaudiffred