Ribbit-Network/ribbit-network-frog-hardware #204: Enclosure: PoE Support for V4 Frog Sensor

keenanjohnson, 2023-02-03:
In order to support a test at an Ameriflux monitoring station, we need to create a version of the Frog Sensor that supports Power over Ethernet (PoE).

@damz has been doing good work supporting a [Power over Ethernet module in Software](https://github.com/Ribbit-Network/ribbit-network-frog-software/pull/32), but we will need to modify the enclosure to support this as well.

### Hardware Selection
The PoE Featherwing from Silicognition is the selected hardware that allows data and power over the ethernet connector in a featherwing form-factor. 

This board can be purchased from several places:

* [Amazon](https://www.amazon.com/gp/product/B08KTVD7BR/ref=sw_img_1?smid=A3MSGSZ93QJG1S&psc=1)
* [Mouser](https://www.mouser.com/ProductDetail/Silicognition/CS-POEFEATHER-01?qs=hWgE7mdIu5QddMUXdZnt2A%3D%3D&gclid=CjwKCAiA_vKeBhAdEiwAFb_nrX0lzXUKBNW3-cXWsQ_DCekkhhjqKgbTiD1w3G7DzInxcdu0xdi_4BoChnUQAvD_BwE) 

@keenanjohnson and @damz both have testing modules.

### Enclosure Modification

@damz provided the image below showing that the board will obviously not fit in the current enclosure.

[image]

It seems that the enclosure will need to become deeper to support the added height of the ethernet connector.

The other big decision is whether to turn the microcontroller and PoE hat so that the cable comes out of the bottom or route the cable out of the side of the enclosure using a cable gland.

### Scalability

This design is initially for a one-off test, so I'm not so worried about it being easy to produce or anything.

keenanjohnson, 2023-02-03:
It doesn't seem like there is a cad model available for this board on any sites I could find. I sent a message asking if the vendor could provide us with a 3d model.

keenanjohnson, 2023-02-03:
@spestana just thought of this, but I wanted to confirm that the Windriver site is using a standard RJ-45 ethernet cable right?

keenanjohnson, 2023-02-07:
Let me know if you need specific other dimensions @eaudiffred !

eaudiffred, 2023-02-07:
@keenanjohnson, Thanks for all the info!  Looks good.  I'll be able to dive in this weekend and make the modifications.

eaudiffred, 2023-02-11:
@keenanjohnson, would you be able to take some measurement pictures of the PoE wing and ESP feather connected?  Then I can get an idea of the whole unit dimension.

eaudiffred, 2023-02-11:
@keenanjohnson @spestana 
Here is what I have so far.  Maybe we call this one Bull Frog...It's huge, lol!  Let me know what you think about placement and routing.
https://cad.onshape.com/documents/b3e6eeabf50d585d20f25fc6/w/a4a82302d129f025f23b244a/e/d204eaabda9d0f822641f52b
[image]
[image]
[image]

keenanjohnson, 2023-02-12:
Haha yeah Bull Frog is a great name. Looks good to me in general shape. 

I think the connector hole on the bottom might need to be larger to allow the ethernet cable to fit through?

I've attached some photos of the stackup. The total height seems to be about 30 mm and the length is about 60 mm.

[image]
[image]

eaudiffred, 2023-02-12:
@keenanjohnson, great, thanks for those measurements!  Based on those I was able to shrink it up a little.  I just went ahead and made a separate hole for the ethernet cable.

spestana, 2023-02-24:
Thanks for you work on this @eaudiffred , it looks great. @keenanjohnson do you think we're still on track to send off the hardware by end of March?

keenanjohnson, 2023-02-24:
I believe so! I have all the electronics required here and enough filamint to print an enclosure whenever we are ready! Just let me know when the time is right @eaudiffred!

eaudiffred, 2023-02-25:
@keenanjohnson, Ready for a print.  
[FrogV4_POE.zip](https://github.com/Ribbit-Network/ribbit-network-frog-hardware/files/10831423/FrogV4_POE.zip)

keenanjohnson, 2023-02-26:
Thanks @eaudiffred ! I won't have time to print this for another week or two due to travel, but that should still be plenty of time to print and test before April 1.

keenanjohnson, 2023-03-16:
Starting a test print now!

eaudiffred, 2023-03-22:
Sounds good.  I've added screw holes for at all four posts.

eaudiffred, 2023-03-22:
@keenanjohnson 
[FrogV4_POE.zip](https://github.com/Ribbit-Network/ribbit-network-frog-hardware/files/11045574/FrogV4_POE.zip)

eaudiffred, 2023-03-22:
Have you tried the other components?  Everything is scaled, so functionally it should be the same.

keenanjohnson, 2023-03-24:
Yup everything else looks great :)

eaudiffred, 2023-03-28:
@keenanjohnson, is there anything else we need to complete on this one?

keenanjohnson, 2023-03-28:
Nope! let's close it out! Thanks for your help!

closed 2023-03-28 by keenanjohnson