Ribbit-Network/ribbit-network-frog-hardware #205: Enclosure Mounting: Pole

eaudiffred, 2023-02-07:
Refine the mounting option for a post/pole.

eaudiffred, 2023-02-10:
I think this will work.  I have a final test print running now.
@keenanjohnson,  Linked here:
https://cad.onshape.com/documents/b3e6eeabf50d585d20f25fc6/w/a4a82302d129f025f23b244a/e/bb2bd1948feb925547304752
I'm just using standard zip ties.
[image]
[image]

[image]

keenanjohnson, 2023-02-10:
Looks great @eaudiffred! Let me know how the final test print goes and we can start printing some to include in the Trane kits!

eaudiffred, 2023-02-10:
@keenanjohnson it turned out well!  I started a larger batch this morning.

closed 2023-02-23 by eaudiffred