Ribbit-Network/ribbit-network-frog-hardware #209: USB Right Angle Connectors Don't Always Insert All The Way

keenanjohnson, 2023-03-22:
I noticed when inspecting the sensors from the recent large order that there seems to be some tolerance differences in either the enclosures or the USB right angle connectors. Some of the connectors would not insert fully into the connector. 

I am sure what the right way to solve this is. Perhaps the braces only need to support the connector on the top and right, but not the left side? 

[image]

eaudiffred, 2023-03-28:
This is a tricky one.  I noticed the same thing.  I purchased a few different brands of USBC right angle connectors, and each was slightly different dimensionally.  
We could eliminate the top support.  The right support helps keep it straight.  The left support is doing all the heavy lifting though - the connector is resting on that one as the power cord pulls down.  
We could try and shift it to the left more.  It would sit less on the plastic body and more on the metal, but I think that's ok.  
[image]
We still have the option to go back to the zip-tie, but that adds parts.

eaudiffred, 2023-03-28:
Maybe we should instead release the tension on the power cable form being applied to that point.  If we had some sort of clip on the bottom of the base for the power cord, it could relieve the tension and eliminate the need for the supports inside the base.  
[image]

eaudiffred, 2023-05-02:
I removed the top support and shifted the left support 0.6mm to the left for more clearance.  I also added a cable clip that we can experiment with.
[image]
[image]

josuah, 2023-05-02:
This might also help with robustness: the USB connector is often the first thing to break in any device.
Type-C might have a better rating, in practice, it is still fragile.

Something that laptop does for instance, is a cable plugged to the internal connector (barebone pins) and an actual connector *screwed* to the enclosure.

Some very small cable would eliminate any form of strain onto the soldered connector of the devboard.

josuah, 2023-05-02:
To take the same supplier:
- https://www.adafruit.com/product/4218 - $9.95 - panel mount cable
- https://www.adafruit.com/product/4432 - $3.95 - Angle adapter

The current solution wins by the price.
Though, if changing it, I'd still consider adding plenty of support to limit the strain:
[image]

Good idea though!

eaudiffred, 2023-05-25:
Here is the latest idea I have.  This gets the pulling force off of the adapter.  

[image]
[image]
[image]

keenanjohnson, 2023-05-26:
Wow looks great to me!

josuah, 2023-05-29:
> I also added a cable clip
Now I see better what this means.
That sounds quite robust!

eaudiffred, 2023-05-29:
> 

I think it will relieve any pressure from the USBC connection.  I did order one of those panel mount USBC connectors that you linked above.  I like that idea better than the right angle connector, I'm just worried about space requirements.  It's supposed to show up tomorrow, so we'll see how it fits.

closed 2023-06-13 by eaudiffred