Ribbit-Network/ribbit-network-frog-hardware #210: Sensors Cables Are Easy to Squish When Clipping on the Head

keenanjohnson, 2023-03-22:
When looking at the finished sensors after our workshop, a small number had squished sensor cables as shown which prevented them from working correctly. [image]

muditprotect3d, 2023-04-01:
Making the notch would remove the stress from the lid on those and it wouldn't increase the size of the enclosure either, I think that could be the most optimal solution

eaudiffred, 2023-04-10:
Added two small notches for cable routing.  One notch on the base, one on the shroud.

closed 2023-04-10 by eaudiffred

eaudiffred, 2023-06-02:
Adding a couple pictures to show the results:
[image]
[image]

keenanjohnson, 2023-06-03:
I like it!