Ribbit-Network/ribbit-network-frog-hardware #211: Microcontroller Can Lift Up When Screws are Inserted

keenanjohnson, 2023-03-22:
We only have screws in two of the four holes on the microcontroller which are both located on the side nearest the USB Connector.

This can often result in the microcontroller's opposite side lifting off of the locating pins when the screws are tightened. I noticed this when inspecting some of the units after our last corporate event. 

This is not ideal as it adds strain on the USB connector.

I'm not sure if the right solution is to go back to four screws or explore the slot idea we have previously explored.

muditprotect3d, 2023-04-01:
I can give it a try with the sliding mechanism if we decide that's the solution we want to go with, but from what I remember when we tried this in December one of the issues was fitting the carrier board in too

eaudiffred, 2023-04-28:
What if instead of locator pins we created a backstop with a notch.  Kind of combining the sliding idea and locator idea.  This way we can keep the enclosure the same size and also not have to add more screws.  

[image]

keenanjohnson, 2023-04-28:
That looks like a much better concept to me!

The only issue is that it creates an overhang for the 3D printing. In my
experience we shouldn’t have issues 3D printing it and we could make the
overhang notch rounded instead of right angles to re risk it further.

eaudiffred, 2023-05-02:
Yeah, good point.  Maybe something like this:
[image]

keenanjohnson, 2023-05-02:
Yeah that's exactly what I was thinking!

eaudiffred, 2023-06-02:
Test prints turned out well.  This along with the cable clip reduces the force on the adapter and frees up the stress on the board.

keenanjohnson, 2023-06-03:
Perfect!

closed 2023-06-13 by eaudiffred