Ribbit-Network/ribbit-network-frog-hardware #212: Head Clips Can Sometimes Break Off

keenanjohnson, 2023-03-22:
Several groups assembling Frog Sensors during our last corporate event reported that their head clips broke off during assembly. 

This may be related to poor 3D printing quality for some of our parts in the last order, but perhaps there is also something we should consider to make the design more robust to that?

[image]

eaudiffred, 2023-03-28:
Thanks for getting pictures @keenanjohnson!  I'm thinking some fillet work could help this problem.

muditprotect3d, 2023-04-01:
Agreed @eaudiffred filleting those edges would reduce some of the stress flow coming down on them when they are being forced in

eaudiffred, 2023-04-10:
Added some fillets to the outside of the clips.  This may need a few test prints to fine tune.  
[image]

eaudiffred, 2023-05-20:
Just finished a test print.  This turned out well.  I was able to apply a lot of pressure to the clip tab without it snapping.

keenanjohnson, 2023-05-23:
Sounds awesome @eaudiffred ! I've reopened this just so we can commit the updated parts. Can you attach them here or open a PR with them?

eaudiffred, 2023-05-23:
Yes!  I'll submit a PR with all of the updated V4 parts once they are complete.

closed 2023-06-13 by eaudiffred