Ribbit-Network/ribbit-network-frog-hardware #216: V4 Lid Redesign

eaudiffred, 2023-05-21:
The lid design needs to be redesigned to eliminate the tab and make it easier to take on/off.

eaudiffred, 2023-05-21:
Adding some info @muditprotect3d shared in discord:
[image]

eaudiffred, 2023-05-21:
@muditprotect3d I used that idea and applied it to the frog sensor lid.  I think it works great!  Awesome idea!

[image]

eaudiffred, 2023-05-21:
The test print turned out great!

[image]
[image]
[image]

keenanjohnson, 2023-05-23:
Oh that's really interesting! Seems like it could be a more robust design to tolerance differences in different printers / filemint brands, etc

eaudiffred, 2023-05-26:
After a few test prints I'm pretty happy with the results!  Here are the STL files if anyone wants to try it out.  
[Base_Lid_V4.zip](https://github.com/Ribbit-Network/ribbit-network-frog-hardware/files/11578362/Base_Lid_V4.zip)

keenanjohnson, 2023-05-28:
Cool. I'll do a test print here.

eaudiffred, 2023-05-29:
@keenanjohnson, how did it turn out?  What are your thoughts?

keenanjohnson, 2023-05-29:
I really like the concept and it slides on really well!

I think we might need to add a feature to stop the sliding in one direction though, so it's easy to just slide on the lid until you hit the feature.

My lid slides around all over the place haha. https://github.com/Ribbit-Network/ribbit-network-frog-hardware/assets/2559382/8b23278f-0735-402e-ab14-482474135e9c

eaudiffred, 2023-05-29:
Looks like differences in printer tolerances again.  The few I've printed aren't so free floating.  Thanks for printing a set.  I'll think about how we can incorporate a stop.  Maybe just stopping the tracks on the base.

keenanjohnson, 2023-05-29:
Ah I see. That's the benefit of collaboration between us :)

eaudiffred, 2023-05-29:
A small change, but I think it will be effective in limiting the slide through.  
[image]
[image]
[image]
[image]

keenanjohnson, 2023-05-29:
Looks perfect to me! Like you said, I don't think much is needed there.

eaudiffred, 2023-05-29:
[Base_Lid_V4.zip](https://github.com/Ribbit-Network/ribbit-network-frog-hardware/files/11593401/Base_Lid_V4.zip)
Here are the file if you have time to run a test today.  I'm going to start one now.

keenanjohnson, 2023-05-29:
Will start a test now!

keenanjohnson, 2023-05-30:
Looks much better to me now.

https://github.com/Ribbit-Network/ribbit-network-frog-hardware/assets/2559382/1adf828f-adc4-4f4b-b18c-379c9a8d9724

Lid locates well.

eaudiffred, 2023-05-30:
@keenanjohnson, that video shows the lid still being able to slide freely off of both sides.  Is that previous version?  
I was hoping that the little notch and step would create a point at which the lid could no longer slide to the left.

keenanjohnson, 2023-05-30:
Oh whoops. I guess I didn't take a new video. Anyway, it works great as you would expect.

closed 2023-06-11 by keenanjohnson