Ribbit-Network/ribbit-network-frog-hardware #220: Sliding Lid Tolerance

eaudiffred, 2023-06-13:
Opening this issue to discuss the tolerance and fit of the sliding lid.

eaudiffred, 2023-06-13:
@keenanjohnson, Interesting that the lid and base you printed are so loose!  When I print them on my Ender 3 V2 its a pretty snug fit.  Tight enough that if I push the lid halfway on, I can grab the other half and shake it pretty good without the two separating, lol.  I'm glad we are discovering these differences.

Thanks for assuming I planned the Lid/shroud interference fit, lol!  Totally lucked into that one.  

Would you mind printing another lid test to see if we can tighten up the fit?  I believe HCG uses the same Prusa printer as you.  I can post a slightly modified file in this issue.

keenanjohnson, 2023-06-13:
Sure happy to do so! I wonder if it might also be due to material shrinkage instead of just printer tolerance.

eaudiffred, 2023-06-13:
[LidV4_TighterFit.zip](https://github.com/Ribbit-Network/ribbit-network-frog-hardware/files/11740095/LidV4_TighterFit.zip)

keenanjohnson, 2023-06-13:
Perfect! I'll print one out now!

eaudiffred, 2023-06-13:
@keenanjohnson, good call, very well could be the material.  

The above file is tighter by 0.4mm

keenanjohnson, 2023-06-13:
Yeah I've been doing my test prints with Hatchbox PETG. What have you been testing with @eaudiffred ?

keenanjohnson, 2023-06-14:
The new lid is much better. https://github.com/Ribbit-Network/ribbit-network-frog-hardware/assets/2559382/71cfb5a7-0fde-416c-a2ff-5fe2ab7b59d8

[image]

eaudiffred, 2023-06-14:
@keenanjohnson, I have been using Overture PETG.  

I forgot I put the frog on the front.  That one is a little too deep.  It worked better at a 0.2mm extrude.  That one you have is a 0.6mm and it got a little sloppy.  What do you think?  Unnecessary/overkill, or cool?

eaudiffred, 2023-06-14:
Glad the tolerance is better.  I do want to to have some resistance to stay in place.

keenanjohnson, 2023-06-14:
Thanks! Yeah I think the frog on the front could be nice, but it's a little deep right now for sure :)

closed 2023-06-24 by eaudiffred